limengdu/MR60BHA2_ESPHome_external_components #1: A fatal error occurred: Contents of segment at SHA256 digest offset 0xb0 are not all zero. Refusing to overwrite.

Rudd-O, 2024-12-13:
Trying to build with ESPHome 2024.12.0b1 for a brand new device to be updated, and I get the fatal error in the headline.

Relevant YAML:

```yaml
substitutions:
  name: "baby-vitals-monitor"
  friendly_name: "Baby vitals monitor"

esphome:
  name: "${name}"
  friendly_name: "${friendly_name}"
  project:
    name: "seeedstudio.mr60bha2_kit"
    version: "2.0"
  platformio_options:
    board_upload.maximum_size: 4194304
  min_version: "2024.3.2" # Fix logger compile error on ESP32-C6 esphome#6323

esp32:
  board: esp32-c6-devkitc-1
  variant: esp32c6
  flash_size: 4MB # upload.flash_size
  framework:
    type: esp-idf

```

Rudd-O, 2024-12-13:
This fixed it:

https://github.com/esphome/issues/issues/4917#issuecomment-2404340471

but using 4MB for the sdkconfig option instead of 8MB, and `version: 5.2.2` and `platform_version: 6.7.0`:

```

esp32:
  board: esp32-c6-devkitc-1
  variant: esp32c6
  flash_size: 4MB # upload.flash_size
  framework:
    type: esp-idf
    # Fix for checksum issue.
    version: 5.2.2
    platform_version: 6.7.0    
    sdkconfig_options:
      CONFIG_ESPTOOLPY_FLASHSIZE_4MB: y
```

Unfortunately when I successfully flash this, the device panics on boot with a watchdog error (only visible through serial console).

Rudd-O, 2024-12-13:
Panic:

```
[16:42:22][I][app:029]: Running through setup()...
[16:42:22][C][i2c.idf:017]: Setting up I2C bus...
[16:42:22][I][i2c.idf:252]: Performing I2C bus recovery
[16:42:22][D][esp-idf:000]: I (759) gpio: GPIO[23]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 1| Pulldown: 0| Intr:0 
[16:42:22]
[16:42:22][D][esp-idf:000]: I (760) gpio: GPIO[22]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 1| Pulldown: 0| Intr:0 
[16:42:22]
[16:42:27]E (5758) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
[16:42:27]E (5758) task_wdt:  - loopTask (CPU 0)
[16:42:27]E (5758) task_wdt: Tasks currently running:
[16:42:27]E (5758) task_wdt: CPU 0: loopTask
[16:42:27]E (5758) task_wdt: Aborting.
[16:42:27]
[16:42:27]
[16:42:27]Core  0 register dump:
[16:42:27]MEPC    : 0x42026316  RA      : 0x42027440  SP      : 0x40826b90  GP      : 0x408168d0  
/home/user/.platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-addr2line: '/home/user/Projects/esphome/config/.esphome/build/baby-vitals-monitor/.pioenvs/baby-vitals-monitor/firmware.elf': No such file
/home/user/.platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-addr2line: '/home/user/Projects/esphome/config/.esphome/build/baby-vitals-monitor/.pioenvs/baby-vitals-monitor/firmware.elf': No such file
[16:42:27]TP      : 0x407f8648  T0      : 0x00000000  T1      : 0x3ffffd52  T2      : 0xffffffff  
[16:42:27]S0/FP   : 0x00000000  S1      : 0x408213a0  A0      : 0x00000000  A1      : 0x40821094  
[16:42:27]A2      : 0x00000001  A3      : 0x00000800  A4      : 0x00000013  A5      : 0x60004000  
[16:42:27]A6      : 0x00000004  A7      : 0x00000014  S2      : 0x00000000  S3      : 0x00000098  
[16:42:27]S4      : 0x00000014  S5      : 0x00000084  S6      : 0x000003e8  S7      : 0x00000002  
[16:42:27]S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
[16:42:27]T3      : 0x4081e000  T4      : 0x00000001  T5      : 0x4081e000  T6      : 0x00000001  
[16:42:27]MSTATUS : 0x00000000  MTVEC   : 0x00000000  MCAUSE  : 0x40820f64  MTVAL   : 0x00000001  
[16:42:27]MHARTID : 0x420c4000  
[16:42:27]
[16:42:27]Stack memory:
[16:42:27]40826b90: 0x00000000 0x00000000 0x40820f64 0x00000001 0x420c4000 0x00000000 0x00000001 0x40820b78
[16:42:27]40826bb0: 0x00000000 0x00000000 0x40820f64 0x4200a954 0x00000000 0x00000001 0x00000006 0x00000078
[16:42:27]40826bd0: 0x40820b7c 0x40820b78 0x00000012 0x4200aad0 0x00000001 0x4081e4f0 0x40816210 0x421266e4
[16:42:27]40826bf0: 0x420c4000 0x00000008 0x420b3000 0x40820b8c 0x00000001 0x420b1000 0x40820b78 0x4200adda
[16:42:27]40826c10: 0x4081984c 0x00000001 0x00000016 0x00000017 0x00000101 0x0000c350 0x00000000 0x00000000
[16:42:27]40826c30: 0x00000001 0x4082126c 0x4081984c 0x42016836 0x00ffffff 0x00000009 0x00000009 0x40820f54
[16:42:27]40826c50: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[16:42:27]40826c70: 0x00000000 0x00000000 0x00000000 0x42008188 0x00000000 0x00000000 0x00000000 0x00000000
[16:42:27]40826c90: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
[16:42:27]40826cb0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000150 0x40826b10 0x00000098 0x4081a204 0x4081a204
[16:42:27]40826cd0: 0x40826cc0 0x4081a1fc 0x00000018 0x40821494 0x40821494 0x40826cc0 0x00000000 0x00000001
[16:42:27]40826cf0: 0x40824cbc 0x706f6f6c 0x6b736154 0x00000000 0x00000000 0x40826cb0 0x00000001 0x00000001
[16:42:27]40826d10: 0x00000000 0x00000000 0x00000000 0x4081f23c 0x4081f2a4 0x4081f30c 0x00000000 0x00000000
[16:42:27]40826d30: 0x00000001 0x00000000 0x00000000 0x00000000 0x42044034 0x00000000 0x00000000 0x00000000
[16:42:27]40826d50: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[16:42:27]40826d70: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[16:42:27]40826d90: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[16:42:27]40826db0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[16:42:27]40826dd0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[16:42:27]40826df0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[16:42:27]40826e10: 0x0000000c 0x0c1a0008 0x00000000 0x40820f54 0x0000000c 0x00000000 0x4081e834 0x40826e1c
[16:42:27]40826e30: 0x0000000c 0x00000000 0x4081e834 0x40826e2c 0x00000010 0x00000000 0x40826cc0 0x00000000
[16:42:27]40826e50: 0x00000000 0x00000020 0x79626162 0x7469762d 0x2d736c61 0x696e6f6d 0x00726f74 0x79626142
[16:42:27]40826e70: 0x74697620 0x40826e50 0x00000020 0x79626142 0x74697620 0x20736c61 0x696e6f6d 0x00726f74
[16:42:27]40826e90: 0x420c3e10 0x00000002 0x40826e74 0x00000048 0x420c4050 0x00000001 0x7fc00000 0x420b1cdc
[16:42:27]40826eb0: 0x0001c200 0x40826eec 0x00000087 0x00000200 0x00000003 0x00000000 0x00000000 0x00000000
[16:42:27]40826ed0: 0x00000000 0x00000000 0x00000000 0x00000000 0x5b6d3600 0x40826cc0 0x00000204 0x3b305b1b
[16:42:27]40826ef0: 0x5b6d3633 0x655b5d44 0x692d7073 0x303a6664 0x3a5d3030 0x305b1b20 0x6d32333b 0x37282049
[16:42:27]40826f10: 0x20293036 0x6f697067 0x5047203a 0x325b4f49 0x207c5d32 0x75706e49 0x3a6e4574 0x207c3120
[16:42:27]40826f30: 0x7074754f 0x6e457475 0x7c31203a 0x65704f20 0x6172446e 0x203a6e69 0x50207c31 0x756c6c75
[16:42:27]40826f50: 0x31203a70 0x7550207c 0x6f646c6c 0x203a6e77 0x49207c30 0x3a72746e 0x5b1b2030 0x1b0a6d30
[16:42:27]40826f70: 0x006d305b 0x5b006d30 0x00006d30 0xfbe69000 0x0ad63c0f 0x053c2ad0 0x9c4aeaf4 0xedc298e9
[16:42:27]
[16:42:27]
[16:42:27]
[16:42:27]ELF file SHA256: dd365573a
[16:42:27]
[16:42:27]Rebooting...
```

Rudd-O, 2024-12-13:
That also happens even when the 60 GHz device is commented out of ESPHome.  let me comment out the I2C bus stuff...

Rudd-O, 2024-12-13:
Aha!  If I comment out the I2C bus, the device boots normally.

Rudd-O, 2024-12-13:
If I keep the I2C bus code enabled, but I disable scan, the device also boots normally.

Rudd-O, 2024-12-13:
https://github.com/esphome/issues/issues/6365 led me to increase the frequency of scan:

```

i2c:
  sda: GPIO22
  scl: GPIO23
  scan: true
  frequency: 200kHz
  id: bus_a
```

With this, the device no longer crashes.  The illuminance sensor works properly too.  Trying the radar now...

Rudd-O, 2024-12-13:
Yep no crashes with that.  Complete functioning config:

```yaml
substitutions:
  name: "baby-vitals-monitor"
  friendly_name: "Baby vitals monitor"

esphome:
  name: "${name}"
  friendly_name: "${friendly_name}"
  project:
    name: "seeedstudio.mr60bha2_kit"
    version: "2.0"
  platformio_options:
    board_upload.maximum_size: 4194304
  min_version: "2024.3.2" # Fix logger compile error on ESP32-C6 esphome#6323

esp32:
  board: esp32-c6-devkitc-1
  variant: esp32c6
  flash_size: 4MB # upload.flash_size
  framework:
    type: esp-idf
    # Fix for checksum issue.
    version: 5.2.2
    platform_version: 6.7.0    
    sdkconfig_options:
      CONFIG_ESPTOOLPY_FLASHSIZE_4MB: y

# Enable logging
logger:
  hardware_uart: USB_SERIAL_JTAG
  level: DEBUG

# Enable Home Assistant API
api:

ota:
  - platform: esphome

external_components:
  - source:
      type: git
      url: https://github.com/limengdu/MR60BHA2_ESPHome_external_components
      ref: main
    components: [ seeed_mr60bha2 ]
    refresh: 0s

wifi:
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 10.250.1.68
    gateway: 10.250.1.1
    subnet: 255.255.255.0
  reboot_timeout: 120min

# For XIAO ESP32C6 Onboard LED
# light:
#   - platform: status_led
#     name: "Switch state"
#     pin: GPIO15

light:
  - platform: esp32_rmt_led_strip
    id: led_ring
    name: "Light"
    pin: GPIO1
    num_leds: 1
    rmt_channel: 0
    rgb_order: GRB
    chipset: ws2812

i2c:
  sda: GPIO22
  scl: GPIO23
  scan: true
  frequency: 200kHz
  id: bus_a

uart:
  id: uart_bus
  baud_rate: 115200
  rx_pin: 17
  tx_pin: 16
  parity: NONE
  stop_bits: 1

seeed_mr60bha2:
  id: my_seeed_mr60bha2

sensor:
  - platform: bh1750
    name: "Illuminance"
    address: 0x23
    update_interval: 1s
  - platform: seeed_mr60bha2
    breath_rate:
      name: "Real-time respiratory rate"
    heart_rate:
      name: "Real-time heart rate"
    distance:
      name: "Distance to detection object"

button:
  - platform: restart
    name: "Restart"
    disabled_by_default: true
    entity_category: config

  - platform: factory_reset
    name: "Restart with factory default settings"
    id: reset
    disabled_by_default: true
    entity_category: config

  - platform: safe_mode
    name: "Safe mode"
    disabled_by_default: true
    entity_category: config

text_sensor:
  - platform: wifi_info
    ip_address:
      name: "IP address"
      disabled_by_default: true
    ssid:
      name: "Connected SSID"
      disabled_by_default: true
    mac_address:
      name: "MAC address"
      disabled_by_default: true

```

Rudd-O, 2024-12-13:
Bad news: with that config, the heartbeat, distance and breathing sensors all show up as "unknown".

Rudd-O, 2024-12-13:
Reducing frequency to 100 KHz works with the illuminance sensor but not the radar.

Rudd-O, 2024-12-13:
Increasing I2C scan timeout to 12ms cures the watchdog problem, but with/without frequency in the config, the radar simply does not appear, and now the illuminance sensor doesn't work either.

Rudd-O, 2024-12-13:
I have completely disabled I2C again (also disabling the illuminance sensor).  Still no functional radar.  Is the UART config even correct here?  I see no errors in ESPHome setup.

Rudd-O, 2024-12-13:
OK it works now!

Final functioning config:

```yaml
substitutions:
  name: "baby-vitals-monitor"
  friendly_name: "Baby vitals monitor"

esphome:
  name: "${name}"
  friendly_name: "${friendly_name}"
  project:
    name: "seeedstudio.mr60bha2_kit"
    version: "2.0"
  platformio_options:
    board_upload.maximum_size: 4194304
  min_version: "2024.3.2" # Fix logger compile error on ESP32-C6 esphome#6323

esp32:
  board: esp32-c6-devkitc-1
  variant: esp32c6
  flash_size: 4MB # upload.flash_size
  framework:
    type: esp-idf
    # Fix for checksum issue.
    version: 5.2.2
    platform_version: 6.7.0    
    sdkconfig_options:
      CONFIG_ESPTOOLPY_FLASHSIZE_4MB: y

# Enable logging
logger:
  hardware_uart: USB_SERIAL_JTAG
  level: DEBUG

# Enable Home Assistant API
api:
  reboot_timeout: 0s

ota:
  - platform: esphome

wifi:
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 10.250.1.68
    gateway: 10.250.1.1
    subnet: 255.255.255.0
  reboot_timeout: 120min

# For XIAO ESP32C6 Onboard LED
# light:
#   - platform: status_led
#     name: "Switch state"
#     pin: GPIO15

light:
  - platform: esp32_rmt_led_strip
    id: led_ring
    name: "Light"
    pin: GPIO1
    num_leds: 1
    rmt_channel: 0
    rgb_order: GRB
    chipset: ws2812

i2c:
  sda: GPIO22
  scl: GPIO23
  scan: true
  timeout: 12ms
  id: bus_a

uart:
  id: uart_bus
  baud_rate: 115200
  rx_pin: 17
  tx_pin: 16
  parity: NONE
  stop_bits: 1

seeed_mr60bha2:
  id: my_seeed_mr60bha2

sensor:
  - platform: bh1750
    name: "Illuminance"
    address: 0x23
    update_interval: 1s
  - platform: seeed_mr60bha2
    breath_rate:
      name: "Real-time respiratory rate"
    heart_rate:
      name: "Real-time heart rate"
    distance:
      name: "Distance to detection object"

button:
  - platform: restart
    name: "Restart"
    disabled_by_default: true
    entity_category: config

  - platform: factory_reset
    name: "Restart with factory default settings"
    id: reset
    disabled_by_default: true
    entity_category: config

  - platform: safe_mode
    name: "Safe mode"
    disabled_by_default: true
    entity_category: config

text_sensor:
  - platform: wifi_info
    ip_address:
      name: "IP address"
      disabled_by_default: true
    ssid:
      name: "Connected SSID"
      disabled_by_default: true
    mac_address:
      name: "MAC address"
      disabled_by_default: true
```

Rudd-O, 2024-12-13:
Simpler configuration that works, without need for custom versions of IDF:

```yaml
substitutions:
  name: "baby-vitals-monitor"
  friendly_name: "Baby vitals monitor"

esphome:
  name: "${name}"
  friendly_name: "${friendly_name}"
  project:
    name: "seeedstudio.mr60bha2_kit"
    version: "2.0"
  platformio_options:
    board_upload.maximum_size: 4194304
  min_version: "2024.3.2" # Fix logger compile error on ESP32-C6 esphome#6323

esp32:
  board: esp32-c6-devkitc-1
  variant: esp32c6
  flash_size: 4MB # upload.flash_size
  framework:
    type: esp-idf
    # Fix for checksum issue.
    # version: 5.2.2
    # platform_version: 6.7.0    
    sdkconfig_options:
      CONFIG_ESPTOOLPY_FLASHSIZE_4MB: y

# Enable logging
logger:
  hardware_uart: USB_SERIAL_JTAG
  level: DEBUG

# Enable Home Assistant API
api:
  encryption:
    key: "oz5gXQG8LlzvO2X8CH0IxXhlIh+nKHvKy6MuuJCIJ+A="
  reboot_timeout: 0s

ota:
  - platform: esphome
    password: "5ef429bac0bff8fd09478b451dafec5b"

external_components:
  - source:
      type: git
      url: https://github.com/limengdu/MR60BHA2_ESPHome_external_components
      ref: main
    components: [ seeed_mr60bha2 ]
    refresh: 0s

wifi:
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 10.250.1.68
    gateway: 10.250.1.1
    subnet: 255.255.255.0
  reboot_timeout: 120min

# For XIAO ESP32C6 Onboard LED
# light:
#   - platform: status_led
#     name: "Switch state"
#     pin: GPIO15

light:
  - platform: esp32_rmt_led_strip
    id: led_ring
    name: "Light"
    pin: GPIO1
    num_leds: 1
    rmt_channel: 0
    rgb_order: GRB
    chipset: ws2812

#i2c:
#  sda: GPIO22
#  scl: GPIO23
#  scan: true
#  frequency: 200kHz
#  id: bus_a

uart:
  id: uart_bus
  baud_rate: 115200
  rx_pin: 17
  tx_pin: 16
  parity: NONE
  stop_bits: 1

seeed_mr60bha2:
  id: my_seeed_mr60bha2

sensor:
#  - platform: bh1750
#    name: "Illuminance"
#    address: 0x23
#    update_interval: 1s
  - platform: seeed_mr60bha2
    breath_rate:
      name: "Real-time respiratory rate"
    heart_rate:
      name: "Real-time heart rate"
    distance:
      name: "Distance to detection object"

  - platform: uptime
    name: "Uptime"
    entity_category: diagnostic

  # Reports the WiFi signal strength/RSSI in dB
  - platform: wifi_signal
    name: "Wi-Fi signal strength"
    update_interval: 10s
    entity_category: diagnostic

button:
  - platform: restart
    name: "Restart"
    disabled_by_default: true
    entity_category: config

  - platform: factory_reset
    name: "Restart with factory default settings"
    id: reset
    disabled_by_default: true
    entity_category: config

  - platform: safe_mode
    name: "Safe mode"
    disabled_by_default: true
    entity_category: config

text_sensor:
  - platform: wifi_info
    ip_address:
      name: "IP address"
      disabled_by_default: true
    ssid:
      name: "Connected SSID"
      disabled_by_default: true
    mac_address:
      name: "MAC address"
      disabled_by_default: true
```

Why can't the sensor detect above 100 BPM?  My baby has a high heart rate.  Can this be changed in any way?

Rudd-O, 2024-12-13:
Why can't the sensor detect above 100 BPM?  My baby has a high heart rate.  Can this be changed in any way?

EDIT: at some point it did get to 120 BPM, but most of the time it detects 70 BPM, which is not at all the heartbeat of the baby.

Rudd-O, 2024-12-13:
Not getting any distance data no matter what I try.

GrumpyMeow, 2024-12-15:
When i flashed the v2.2 firmware (https://github.com/limengdu/MR60BHA2_ESPHome_external_components/releases/download/V2.2/seeedstudio-mr60bha2-kit-esp32c6.factory.bin) , i also saw some crashes on the serial port.

I also noticed that i must open up the case and flash by using the internal USB-C connector of the Xiao itself, not the exposed USB-C connector.

Rudd-O, 2024-12-18:
Yeah I could only flash via serial using the internal ESP USB type C port.  That's fine I have no issue with that, but that should be documented somewhere as a recovery procedure, to let people know that the externally-visible USB type C port does not work for flashing when the device is put in a state of bootloop.

Rudd-O, 2024-12-18:
Either way the device is not useful to me, with heart rate capped at 100 - 120 bpm, my baby has 130+ heart rate.  Also no distance coming from the sensor.  I wish the verbose debug option logged the decoded packets or problems with decoding the packets coming through the UART from the sensor itself -- currently the verbose option only logs the packets raw, which is useless to diagnose whether the code is doing what is intended or not.

limengdu, 2024-12-19:
Currently the [radar's programme](https://esphome.io/components/seeed_mr60bha2) has been checked by ESPHome and added to ESPHome's official support, stability will be more guaranteed. The distance 0 issue has been confirmed, and a version of the radar's firmware will be updated in the near future, and the 3D body presence function will be added. We hope to meet the needs of as many users as possible, the current heart rate detection range is 60~120bpm, which is the current 60GHz radar's ability range, we are still trying to do better, thank you for testing and providing feedback.

GrumpyMeow, 2024-12-20:
The issue with only able to flash over serial, can be resolved by adding a ‘yield()’ to the main loop. See my fork.

limengdu, 2024-12-24:
Thanks to @GrumpyMeow for the code reference, this has now been fixed.
Please use the latest ESPHome firmware:
https://github.com/limengdu/MR60BHA2_ESPHome_external_components/releases
We will provide firmware for updating the radar as well as a tutorial in the near future.

closed 2024-12-24 by limengdu