limengdu/MR60BHA2_ESPHome_external_components #3: Distance not working any more

GrumpyMeow, 2024-12-16:
I compiled a new firmware using the code and notice that the distance is not being set any more. I think there is an issue which expects the data to be of length 8, but i don't see that in the very-verbose output.

Frames of type "0x0a16" are for "distance".
Frames of type "0x0a14" and "0x0a15"  are for respitory/heart-rate.

```
Received Frame: ID: 0x3a1d, Type: 0x0a16, Data: [00.00.00.00] Raw Data: [01.3A.1D.00.04.0A.16.C1.00.00.00.00.FF (13)]
Received Frame: ID: 0x3a29, Type: 0x0a16, Data: [00.00.00.00] Raw Data: [01.3A.29.00.04.0A.16.F5.00.00.00.00.FF (13)]
Received Frame: ID: 0x3a33, Type: 0x0a14, Data: [00.00.00.00] Raw Data: [01.3A.33.00.04.0A.14.ED.00.00.00.00.FF (13)]
Received Frame: ID: 0x3a34, Type: 0x0a15, Data: [00.00.00.00] Raw Data: [01.3A.34.00.04.0A.15.EB.00.00.00.00.FF (13)]
Received Frame: ID: 0x3a35, Type: 0x0a16, Data: [00.00.00.00] Raw Data: [01.3A.35.00.04.0A.16.E9.00.00.00.00.FF (13)]
Received Frame: ID: 0x3a41, Type: 0x0a16, Data: [00.00.00.00] Raw Data: [01.3A.41.00.04.0A.16.9D.00.00.00.00.FF (13)]
Received Frame: ID: 0x3a42, Type: 0x0a14, Data: [00.00.00.00] Raw Data: [01.3A.42.00.04.0A.14.9C.00.00.00.00.FF (13)]
Received Frame: ID: 0x3a43, Type: 0x0a15, Data: [00.00.00.00] Raw Data: [01.3A.43.00.04.0A.15.9C.00.00.00.00.FF (13)]
Received Frame: ID: 0x3a44, Type: 0x0a16, Data: [00.00.00.00] Raw Data: [01.3A.44.00.04.0A.16.98.00.00.00.00.FF (13)]
Received Frame: ID: 0x3a45, Type: 0x0a14, Data: [00.00.00.00] Raw Data: [01.3A.45.00.04.0A.14.9B.00.00.00.00.FF (13)]
```

limengdu, 2024-12-19:
The issue has been confirmed and the firmware for the radar will be updated with a version in the near future. Please wait as the radar's firmware will be synchronised after the update.

limengdu, 2024-12-24:
Thanks for the code reference, this has now been fixed.
Please use the latest ESPHome firmware:
https://github.com/limengdu/MR60BHA2_ESPHome_external_components/releases
We will provide firmware for updating the radar as well as a tutorial in the near future.

closed 2024-12-24 by limengdu