limengdu/MR60BHA2_ESPHome_external_components #5: Release v3.0 confusing or wrong

GrumpyMeow, 2024-12-29:
Hi!
I've tried to upgrade my device to have the newly added presence detection. I don't see this newly added feature working yet.
By retracing my steps i get confused about the naming which seems to be about a different product.

I get confused by the naming of the v3.0 release: `[MR60FDA2 mmWave Kit Firmware V3.0]` which is linked to release `v3.1`.
But also is about a different device: `MR60FDA2` while the files reference the product `seeedstudio-mr60bha2-kit-esp32c6.elf`

bobvandevijver, 2025-01-04:
It seems that you also need to flash the firmware on the mr60bha2 itself, which is described here: https://wiki.seeedstudio.com/getting_started_with_mr60bha2_mmwave_kit/. The V3 here is only for the ESPHome component that is running on the ESP, which uses UART to communicate with the mmWave board itself.

limengdu, 2025-01-17:
Sorry it was a naming error on my part, it's fixed over now. This repository applies to the MR60BHA2 radar, the firmware for this radar is all compiled and generated by esphome automation, the naming error will not affect the normal use of the firmware, please rest assured.