limengdu/MR60BHA2_ESPHome_external_components #6: Unstable/unreliable sensor

bobvandevijver, 2025-01-04:
The sensor is quite unstable/unreliable: it either reboots very quickly in quick succession, or the mmWave module stops reporting if it didn't reboot. For the latter situation only a power cycle works to revive the module.

I've updated the mmWave module firmware to 1.6.4 as described here: https://wiki.seeedstudio.com/getting_started_with_mr60bha2_mmwave_kit/#module-firmware-upgrade.

My current ESPHome configuration:

```
esphome:
  name: "presence-mmwave-bijkeuken"
  friendly_name: Presence mmWave Bijkeuken
  name_add_mac_suffix: false
  platformio_options:
    board_upload.maximum_size: 4194304
  min_version: "2024.3.2" # Fix logger compile error on ESP32-C6 esphome#6323

esp32:
  board: esp32-c6-devkitc-1
  variant: esp32c6
  flash_size: 4MB # upload.flash_size
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_ESPTOOLPY_FLASHSIZE_4MB: y

# Enable logging
logger:
  hardware_uart: USB_SERIAL_JTAG

# Enable Home Assistant API
api:
  reboot_timeout: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true

ota:
  - platform: esphome
    password: !secret ota_password

light:
  - platform: esp32_rmt_led_strip
    id: led_ring
    name: "RGB Light"
    pin: GPIO1
    num_leds: 1
    rmt_channel: 0
    rgb_order: GRB
    chipset: ws2812

i2c:
  sda: GPIO22
  scl: GPIO23
  id: bus_a
  scan: false

uart:
  id: uart_bus
  baud_rate: 115200
  rx_pin: 17
  tx_pin: 16
  parity: NONE
  stop_bits: 1

external_components:
  - source:
      type: git
      url: https://github.com/limengdu/MR60BHA2_ESPHome_external_components
      ref: main
    components: [ seeed_mr60bha2 ]
    refresh: 0s

seeed_mr60bha2:
  id: my_seeed_mr60bha2

binary_sensor:
  - platform: seeed_mr60bha2
    people_exist:
      name: "Occupancy"

sensor:
  - platform: bh1750
    name: "Illuminance"
    address: 0x23
    update_interval: 10s
  - platform: seeed_mr60bha2
    breath_rate:
      name: "Real-time respiratory rate"
    heart_rate:
      name: "Real-time heart rate"
    distance:
      name: "Distance to detection object"
    target_num:
      name: "Person count"
  - platform: uptime
    name: "Uptime"
    entity_category: diagnostic
  - platform: wifi_signal
    name: "WiFi RSSI"
    update_interval: 300s
    id: sensor_wifi_signal

button:
  - platform: restart
    name: "Restart"
    disabled_by_default: true
    entity_category: config

  - platform: factory_reset
    name: "Restart with factory default settings"
    id: reset
    disabled_by_default: true
    entity_category: config

  - platform: safe_mode
    name: "Safe mode"
    disabled_by_default: true
    entity_category: config
```

Is there maybe anything I'm missing?

limengdu, 2025-01-17:
Hello, the firmware for the radar will be updated with a version this week (at the latest) next week, so keep an eye on the wiki to update the radar firmware at that time. This version of the firmware will update the radar's respiratory heart rate stability as well as the frequency of distance reporting.

limengdu, 2025-01-17:
https://github.com/Seeed-Studio/wiki-documents/pull/2119#pullrequestreview-2558279458
The firmware for the radar has been updated.The firmware for the XIAO has also been updated.

closed 2025-01-17 by limengdu

bobvandevijver, 2025-01-18:
@limengdu I'm sorry, but even with 1.6.5 on the module and the latest version of the component it still is completely unreliable. It seems to get stuck in a state where it only reports the last distance to object value, as that is the only value being updated multiple times a second (visible in the esp logs). It keeps detecting 1 person in this state, even though nobody has been in the room for multiple minutes. 

If you do enter, it first just to undetected, to jump back to detected within a couple of seconds.

bobvandevijver, 2025-01-26:
@limengdu See some logging snippets below. Most curious are that it is sending multiple values for the person count at the same time, and that it is getting stuck at some time where it only sends the distance to object detection.

```
[D][sensor:093]: 'Distance to detection object': Sending state 63.14000 cm with 2 decimals of accuracy
[D][sensor:093]: 'Real-time respiratory rate': Sending state 12.00000  with 0 decimals of accuracy
[D][sensor:093]: 'Person count': Sending state 1.00000  with 0 decimals of accuracy
[D][binary_sensor:036]: 'Occupancy': Sending state OFF
[D][sensor:093]: 'Real-time respiratory rate': Sending state 0.00000  with 0 decimals of accuracy
[D][sensor:093]: 'Real-time heart rate': Sending state 0.00000 bpm with 0 decimals of accuracy
[D][sensor:093]: 'Distance to detection object': Sending state 0.00000 cm with 2 decimals of accuracy
[D][sensor:093]: 'Person count': Sending state 0.00000  with 0 decimals of accuracy
[D][light:036]: 'RGB Light' Setting:
[D][light:055]:   Color brightness: 100%
[D][light:058]:   Red: 100%, Green: 0%, Blue: 0%
[D][light:085]:   Transition length: 1.0s
[D][sensor:093]: 'Person count': Sending state 1.00000  with 0 decimals of accuracy
[D][sensor:093]: 'Real-time respiratory rate': Sending state 0.00000  with 0 decimals of accuracy
[D][sensor:093]: 'Real-time heart rate': Sending state 0.00000 bpm with 0 decimals of accuracy
[D][sensor:093]: 'Distance to detection object': Sending state 0.00000 cm with 2 decimals of accuracy
[D][sensor:093]: 'Person count': Sending state 0.00000  with 0 decimals of accuracy
[D][sensor:093]: 'Person count': Sending state 1.00000  with 0 decimals of accuracy
[D][sensor:093]: 'Real-time respiratory rate': Sending state 0.00000  with 0 decimals of accuracy
[D][sensor:093]: 'Real-time heart rate': Sending state 0.00000 bpm with 0 decimals of accuracy
[D][sensor:093]: 'Distance to detection object': Sending state 0.00000 cm with 2 decimals of accuracy
[D][sensor:093]: 'Person count': Sending state 0.00000  with 0 decimals of accuracy
[D][sensor:093]: 'Person count': Sending state 1.00000  with 0 decimals of accuracy
```

```
[D][sensor:093]: 'Real-time respiratory rate': Sending state 0.00000  with 0 decimals of accuracy
[D][sensor:093]: 'Real-time heart rate': Sending state 0.00000 bpm with 0 decimals of accuracy
[D][sensor:093]: 'Distance to detection object': Sending state 0.00000 cm with 2 decimals of accuracy
[D][sensor:093]: 'Person count': Sending state 0.00000  with 0 decimals of accuracy
[D][sensor:093]: 'Person count': Sending state 2.00000  with 0 decimals of accuracy
```

```
[D][sensor:093]: 'Person count': Sending state 0.00000  with 0 decimals of accuracy
[D][sensor:093]: 'Real-time respiratory rate': Sending state 0.00000  with 0 decimals of accuracy
[D][sensor:093]: 'Real-time heart rate': Sending state 0.00000 bpm with 0 decimals of accuracy
[D][sensor:093]: 'Distance to detection object': Sending state 0.00000 cm with 2 decimals of accuracy
[D][sensor:093]: 'Person count': Sending state 0.00000  with 0 decimals of accuracy
[D][sensor:093]: 'Real-time respiratory rate': Sending state 0.00000  with 0 decimals of accuracy
[D][sensor:093]: 'Real-time heart rate': Sending state 0.00000 bpm with 0 decimals of accuracy
[D][sensor:093]: 'Distance to detection object': Sending state 0.00000 cm with 2 decimals of accuracy
[D][sensor:093]: 'Person count': Sending state 0.00000  with 0 decimals of accuracy
[D][sensor:093]: 'Person count': Sending state 1.00000  with 0 decimals of accuracy
[D][sensor:093]: 'Distance to detection object': Sending state 40.18000 cm with 2 decimals of accuracy
[D][binary_sensor:036]: 'Occupancy': Sending state ON
[D][sensor:093]: 'Real-time respiratory rate': Sending state 12.00000  with 0 decimals of accuracy
[D][light:036]: 'RGB Light' Setting:
[D][light:047]:   State: ON
[D][light:051]:   Brightness: 25%
[D][light:055]:   Color brightness: 100%
[D][light:058]:   Red: 0%, Green: 100%, Blue: 16%
[D][light:085]:   Transition length: 1.0s
[D][sensor:093]: 'Distance to detection object': Sending state 40.18000 cm with 2 decimals of accuracy
[D][sensor:093]: 'Real-time respiratory rate': Sending state 12.00000  with 0 decimals of accuracy
[D][sensor:093]: 'Distance to detection object': Sending state 40.18000 cm with 2 decimals of accuracy
[D][sensor:093]: 'Real-time respiratory rate': Sending state 12.00000  with 0 decimals of accuracy
[D][sensor:093]: 'Distance to detection object': Sending state 40.18000 cm with 2 decimals of accuracy
```

```
[D][sensor:093]: 'Distance to detection object': Sending state 172.20000 cm with 2 decimals of accuracy
[D][sensor:093]: 'Distance to detection object': Sending state 172.20000 cm with 2 decimals of accuracy
[D][sensor:093]: 'Distance to detection object': Sending state 172.20000 cm with 2 decimals of accuracy
[D][sensor:093]: 'Distance to detection object': Sending state 172.20000 cm with 2 decimals of accuracy
[D][sensor:093]: 'Distance to detection object': Sending state 172.20000 cm with 2 decimals of accuracy
[D][sensor:093]: 'Distance to detection object': Sending state 172.20000 cm with 2 decimals of accuracy
[D][sensor:093]: 'Distance to detection object': Sending state 172.20000 cm with 2 decimals of accuracy
[D][sensor:093]: 'Distance to detection object': Sending state 172.20000 cm with 2 decimals of accuracy
[D][sensor:093]: 'Distance to detection object': Sending state 172.20000 cm with 2 decimals of accuracy
[D][sensor:093]: 'Distance to detection object': Sending state 172.20000 cm with 2 decimals of accuracy
```