limengdu/MR60BHA2_ESPHome_external_components #8: Boot loop when connecting to wifi

jhansche, 2025-01-28:
I installed the firmware from [here](https://wiki.seeedstudio.com/ha_with_mr60bha2/#step-2-preparing-the-kit), and after entering my wifi information, it then enters a boot loop.

When I try to adopt a new yaml config following the template [here](https://wiki.seeedstudio.com/ha_with_mr60bha2/#modifying-the-firmware-with-esphome), I get a compile error:
```
src/esphome/components/seeed_mr60bha2/seeed_mr60bha2.cpp: In member function 'void esphome::seeed_mr60bha2::MR60BHA2Component::process_frame_(uint16_t, uint16_t, const uint8_t*, size_t)':
src/esphome/components/seeed_mr60bha2/seeed_mr60bha2.cpp:154:17: error: 'class esphome::seeed_mr60bha2::MR60BHA2Component' has no member named 'people_exist_binary_sensor_'
  154 |       if (this->people_exist_binary_sensor_ != nullptr && length >= 2) {
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/seeed_mr60bha2/seeed_mr60bha2.cpp:156:15: error: 'class esphome::seeed_mr60bha2::MR60BHA2Component' has no member named 'people_exist_binary_sensor_'
  156 |         this->people_exist_binary_sensor_->publish_state(people_exist_int);
      |               ^~~~~~~~~~~~~~~~~~~~~~~~~~~
```
which appears to be fixed by #7 (because I had commented out the binary_sensor in an effort to debug the boot loop). When I uncomment that, and comment out my `wifi` configuration, then the firmware compiles successfully and installs, and I can see the state updates getting logged. However, once I add my wifi configuration back, it goes right back into the boot loop.

I will try the changes in #7 to see if that helps.

Note that this was done by connecting to the smaller ESP32-C6 module's USB port (the one that is NOT exposed through the 3D printed case). If I try to connect to the port that is exposed through the case, I do not get a USB serial port that can be used to install the firmware. The only way I can get a USB serial terminal is on the smaller board.

jhansche, 2025-01-29:
It looks like this was a power issue. I was able to flash the firmware connect to my computer once, with wifi, but then I have to connect it to a different power source in order for it to boot. Apparently my computer (macbook pro) is not supplying enough power through the ESP32-C6 to allow wifi to work.

jhansche, 2025-01-29:
I guess I spoke too soon. Using a better power source worked for a while, but it eventually started the same boot loop again.

limengdu, 2025-02-05:
The name of this variable has changed, but I haven't had the problem you're feeding back on my end. Suggest you update ESPHome and try again.

jhansche, 2025-02-05:
I updated to the #7 version and still saw the issue. I know which variable you're talking about. But that is not the issue

limengdu, 2025-02-05:
I think I've found what the problem is, please wait while I fix it.

limengdu, 2025-02-05:
Fixed. Please try again.

closed 2025-02-25 by limengdu

jhansche, 2025-04-16:
I was able to capture the crash. It seems that it happens as soon as wifi finishes connecting, and the i2c bus scan is performed:

<details>
<summary>Logs</summary>

```
[01:52:45][I][wifi:617]: WiFi Connected!
[01:52:45][I][esp-idf:000]: I (5526) mdns_mem: mDNS task will be created from internal RAM
[01:52:45]
[01:52:45][I][app:062]: setup() finished successfully!
[01:52:45][W][component:170]: Component wifi cleared Warning flag
[01:52:45][W][component:157]: Component api set Warning flag: unspecified
[01:52:45][I][app:100]: ESPHome version 2025.3.3 compiled on Apr 16 2025, 00:59:08
[01:52:45][I][app:102]: Project seeedstudio.mr60bha2_kit version 3.3
[01:52:45][I][i2c.idf:102]: Results from i2c bus scan:
[01:52:45][I][i2c.idf:108]: Found i2c device at address 0x23
[01:52:45]Guru Meditation Error: Core  0 panic'ed (Store access fault). Exception was unhandled.
[01:52:45]
[01:52:45]Core  0 register dump:
[01:52:45]MEPC    : 0x42010324  RA      : 0x4200fdaa  SP      : 0x40824e10  GP      : 0x40817400  
[01:52:45]TP      : 0x407f7894  T0      : 0x40022494  T1      : 0x00000000  T2      : 0x73696f4e  
[01:52:45]S0/FP   : 0x40821c48  S1      : 0x00000000  A0      : 0x00000000  A1      : 0x00000000  
[01:52:45]A2      : 0x00000000  A3      : 0x40832074  A4      : 0x42006bbe  A5      : 0x408342a4  
[01:52:45]A6      : 0x0000009a  A7      : 0x00000002  S2      : 0x40834800  S3      : 0x40821c48  
[01:52:45]S4      : 0x40834800  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
[01:52:45]S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
[01:52:45]T3      : 0x00000000  T4      : 0x435f3931  T5      : 0x3535325f  T6      : 0x306b7370  
[01:52:45]MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000007  MTVAL   : 0x00000028  
[01:52:45]MHARTID : 0x00000000  
[01:52:45]
[01:52:45]Stack memory:
[01:52:45]40824e10: 0x40834800 0x00000000 0x40821c48 0x4200fdaa 0x00000000 0x40821c70 0x40821c48 0x4200ff66
[01:52:45]40824e30: 0x40834800 0x40821c70 0x40821c48 0x420101be 0x00000000 0x00000000 0x40834800 0xff015a24
[01:52:45]40824e50: 0x408347fc 0x00000102 0x4081a31c 0x42014224 0x40821768 0x40821760 0x00001603 0x40821c48
[01:52:45]40824e70: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x42007e9a
[01:52:45]40824e90: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[01:52:45]40824eb0: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000154
[01:52:45]40824ed0: 0x40824d60 0x00001382 0x4081ae4c 0x4081ae4c 0x40824ed0 0x4081ae44 0x00000018 0x40832bc8
[01:52:45]40824ef0: 0x40832bc8 0x40824ed0 0x00000000 0x00000001 0x40822ecc 0x706f6f6c 0x6b736154 0x42007e00
[01:52:45]40824f10: 0x00000000 0x00000000 0x40824ec0 0x00000001 0x00000000 0x408292c4 0x420469d0 0x0000000b
[01:52:45]40824f30: 0x4081fd34 0x4081fd9c 0x4081fe04 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000
[01:52:45]40824f50: 0x00000000 0x4209dbb4 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[01:52:45]40824f70: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[01:52:45]40824f90: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[01:52:45]40824fb0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[01:52:45]40824fd0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[01:52:45]40824ff0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[01:52:45]40825010: 0x00000000 0x00000000 0x00000000 0x00000000 0x40000000 0x0000002c 0x00000001 0x40825138
[01:52:45]40825030: 0x40825918 0x00000000 0x40825040 0x00000004 0x69666977 0x00000000 0x40825074 0x4081f434
[01:52:45]40825050: 0x00000003 0x0000000c 0x00000000 0x4081f434 0x40825050 0x00000010 0x00000000 0x40824ed0
[01:52:45]40825070: 0x00000000 0x00000000 0x0000000c 0x0c160008 0x00000000 0x40821d30 0x0000000c 0x00000000
[01:52:45]40825090: 0x4081f434 0x40825084 0x00000034 0x65656573 0x75747364 0x2d6f6964 0x3036726d 0x32616862
[01:52:45]408250b0: 0x74696b2d 0x3531342d 0x00633563 0x408250c4 0x00000004 0x69666977 0x00000000 0x40825094
[01:52:45]408250d0: 0x00000034 0x65656553 0x74532064 0x6f696475 0x36524d20 0x41484230 0x694b2032 0x31342074
[01:52:45]408250f0: 0x63356335 0x00000000 0x00000000 0x0000000c 0x0c160008 0x408250cc 0x00000064 0x420b1fa4
[01:52:45]40825110: 0x00000002 0x7fc00000 0x420b1f94 0x0001c200 0x40825174 0x0000003d 0x00000200 0x00000003
[01:52:45]40825130: 0x00000000 0x74696b2d 0x00000000 0x40825028 0x40825918 0x40825028 0x00000002 0x40833164
[01:52:45]40825150: 0x40833174 0x40833174 0x00000003 0x74532000 0x40824ed0 0x00000000 0x00000000 0x00000000
[01:52:45]40825170: 0x00000204 0x3b305b1b 0x5b6d3233 0x695b5d49 0x692e6332 0x313a6664 0x3a5d3830 0x756f4620
[01:52:45]40825190: 0x6920646e 0x64206332 0x63697665 0x74612065 0x64646120 0x73736572 0x32783020 0x305b1b33
[01:52:45]408251b0: 0x2e33006d 0x305b1b33 0x3020006d 0x39353a30 0x1b38303a 0x006d305b 0x6e726574 0x52206c61
[01:52:45]408251d0: 0x5b1b4d41 0x1b0a6d30 0x006d305b 0x202c3035 0x6b73616d 0x3532203a 0x35322e35 0x35322e35
[01:52:45]408251f0: 0x2c302e34 0x3a776720 0x32393120 0x3836312e 0x2e30392e 0x305b1b31 0x5b1b0a6d 0x5b006d30
[01:52:45]
[01:52:45]
[01:52:45]
[01:52:45]ELF file SHA256: aafe4bc7174d6234
[01:52:45]
[01:52:45]Rebooting...
```
</detail>

Which is interesting, because I tried to remove all references of wifi and other nonessential configurations from my yaml file, and the installer reported that it was successful. But obviously this is still trying to connect to my wifi network, and it's still scanning the i2c bus.

@limengdu Does this panic crash look familiar?

limengdu, 2025-04-16:
The firmware has not made any updates to the light functions, and the cause of instability requires further analysis. The radar firmware update relies on software provided by the original chip manufacturer, which currently only has a Windows version, and we have no other alternatives.

limengdu, 2025-04-16:
I have not encountered the information in your log
Have you modified the YAML code?

jhansche, 2025-04-16:
Yes, I've tried commenting out parts until I can get it to run stably. So far the only thing that seems to prevent the boot loop is disabling wifi.

I have a second one that has been working relatively well after the first few days of messing with it, and the only difference in that one is I added a bunch of `filters` to reduce the load on the ha core. But I think there's more going on here than just that.