limengdu/MR60BHA2_ESPHome_external_components #9: ESPHome RMT Breaking Change

GuyFromMars365, 2025-02-21:
Need help modifying your mr60bha2.yaml example, following ESPHome breaking change. 

https://github.com/esphome/esphome/pull/7770

**LOG:**

`INFO ESPHome 2025.2.0
INFO Reading configuration /config/esphome/mmwv-rabbit-room.yaml...
INFO Updating https://github.com/limengdu/MR60BHA2_ESPHome_external_components@main
Failed config

light.esp32_rmt_led_strip: [source /config/esphome/mmwv-rabbit-room.yaml:64]
  platform: esp32_rmt_led_strip
  id: led_ring
  name: Seeed MR60BHA2 RGB Light
  pin: GPIO1
  num_leds: 1
  
  This feature is not available for the IDF framework version 5.
  rmt_channel: 0
  rgb_order: GRB
  chipset: ws2812`

robidog, 2025-02-23:
I marked out "rmt_channel: 0" on line 63

limengdu, 2025-02-25:
This issue may require the person who wrote the ws2812 to assist in updating the ws2812 code

closed 2025-02-25 by limengdu

GuyFromMars365, 2025-02-25:
@limengdu that doesn't answer my question. However, @robidog is correct, commenting out "rmt_channel: 0" allows me to sucessfully compile an update. ...I'm not familiar enough with ESPHome yet to debug or determine what function rmt_channel serves. Do you know? Will commenting this out render the LED inoperative?

robidog, 2025-02-25:
@GuyFromMars365 yes in fact the LED still worked for me after commenting out the line.