limengdu/MR60BHA2_ESPHome_external_components #11: Cannot compile with latest 2025.2.2 ESPHome

Rudd-O, 2025-03-09:
With or without

```yaml
external_components:
  - source:
      type: git
      url: https://github.com/limengdu/MR60BHA2_ESPHome_external_components
      ref: main
    components: [ seeed_mr60bha2 ]
    refresh: 0s
```

I get

```
src/esphome/components/seeed_mr60bha2/seeed_mr60bha2.cpp:132:13: error: 'class esphome::seeed_mr60bha2::MR60BHA2Component' has no member named 'has_target_binary_sensor_'
```

Rudd-O, 2025-03-09:
Related: https://github.com/esphome/issues/issues/6108

Workaround:

```yaml
binary_sensor:
  - platform: seeed_mr60bha2
    has_target:
      name: "Presence Information"
````

limengdu, 2025-03-10:
https://github.com/limengdu/MR60BHA2_ESPHome_external_components/blob/f875eba14198a61f13d39182ba6c2558b7401511/example/mr60bha2.yaml#L11

closed 2025-03-10 by limengdu

Rudd-O, 2025-03-15:
awesome bro, when is this hitting upstream?  do we have a pr open?

Rudd-O, 2025-03-15:
I tried your latest branch as external component and it didn't fix the issue AT ALL.  please reopen.