limengdu/MR60BHA2_ESPHome_external_components #12: Doesn't work at all now

Rudd-O, 2025-03-15:
I have compiled a firmware using this component as external:

```
external_components:
  - source: github://limengdu/MR60BHA2_ESPHome_external_components
    components: [seeed_mr60bha2]

uart:
  id: uart_bus
  baud_rate: 115200
  rx_pin: 17
  tx_pin: 16
  parity: NONE
  stop_bits: 1

seeed_mr60bha2:
  id: my_seeed_mr60bha2

binary_sensor:
  - platform: seeed_mr60bha2
    has_target:
      name: "Target detected"

sensor:
  - platform: seeed_mr60bha2
    breath_rate:
      name: "Real-time respiratory rate"
    heart_rate:
      name: "Real-time heart rate"
    distance:
      name: "Distance to detection object"
```

Now none of the MR60BHA2 sensors work (they used to work although the device crashed a lot).

They all say Unknown in Home Assistant.

All other non MR60BHA2 sensors work correctly.

What happened?

Rudd-O, 2025-03-15:
```
`[11:04:04][E][seeed_mr60bha2:115]: DATA_CKSUM_FRAME ERROR: 0x48
```

Sometimes.

Rudd-O, 2025-03-15:
If I disable the distance to detection object sensor, suddenly the breath and heart rate sensors begin working.

ragharamani, 2025-03-15:
I'm having the same issue,  the luminance sensor works but nothing else does. even when I disable "Distance to detection object"

`i2c:
  sda: GPIO22
  scl: GPIO23
  scan: true
  id: bus_a

uart:
  id: uart_bus
  baud_rate: 115200
  rx_pin: 17
  tx_pin: 16
  parity: NONE
  stop_bits: 1

seeed_mr60bha2:
  id: my_seeed_mr60bha2

binary_sensor:
  - platform: seeed_mr60bha2
    has_target:
      name: "Person Information"

sensor:
  - platform: bh1750
    name: "Seeed MR60BHA2 Illuminance"
    address: 0x23
    update_interval: 1s
  - platform: seeed_mr60bha2
    breath_rate:
      name: "Real-time respiratory rate"
    heart_rate:
      name: "Real-time heart rate"
    #distance:
      #name: "Distance to detection object"
    num_targets:
      name: "Target Number"
`

dsoiM, 2025-03-16:
> I'm having the same issue, the luminance sensor works but nothing else does. even when I disable "Distance to detection object"
>
> ```
> i2c:
>   sda: GPIO22
>   scl: GPIO23
>   scan: true
>   id: bus_a [...]

Same issue here. I used the prebuilt firmware from https://limengdu.github.io/MR60BHA2_ESPHome_external_components/

koszta, 2025-03-17:
It's likely due to some recent changes. I reverted all the commits that were blocking sensor data to be directly sent to HA to see what is actually received from the radar module. Now it seems like it's sending sensor data. It's still not super stable using the radar firmware version 1.6.4, I'll try later to flash 1.6.5.

https://github.com/koszta/MR60BHA2_ESPHome_external_components/tree/restore-publishing-all-states

You can try it by replacing the git url:

```yaml
external_components:
  - source:
      type: git
      url: https://github.com/koszta/MR60BHA2_ESPHome_external_components
      ref: restore-publishing-all-states
    components: [ seeed_mr60bha2 ]
    refresh: 0s
```

Not sending all the data from the sensor is discouraged, see the comments of an ESPHome developer:
https://github.com/esphome/esphome/pull/8209#issuecomment-2646731542

koszta, 2025-03-18:
If you don't want to send duplicate updates, you can use the ESPHome filters for example

```yaml
  distance:
      name: "Distance to detection object"
      filters:
        - delta: 0.01
```

koszta, 2025-03-19:
With the help of the reset pin (GPIO21) you can reset the stuck radar module, so you don't need to power cycle it, you can do it remotely. It should be possible to trigger this on certain conditions or periodically.

```yaml
output:
  - platform: gpio
    pin: GPIO21
    inverted: True
    id: reset

button:
  - platform: output
    name: "Reset radar"
    output: reset
    duration: 50ms
```

koszta, 2025-03-20:
The latest released radar firmware (1.6.4) is really unreliable, unfortunately there is no way to make this radar working without a fix for that.

In this repository there are some workarounds were applied in the C++ code to make the values look more reliable, but in fact it just made things worse.

So I forked this project and changed the following:

- Removed all the hacks from the C++ code that was zeroing other sensors when there was no person detected (one of the least reliable sensors).
- Removed the hacks to prevent the same value being sent out from C++ and replaced them with sensor filters. It is an ESPHome anti-pattern, it's better to publish everything that comes from the sensor, so you can easily comment out the filters in the YAML file to see what is going on. It's great for debugging as flashing your board will not reset the radar.
- Added a timeout filter to the sensors, so if there has been no value received for a while, it will set the fields to 'Unknown'.
- Added a "Reset radar" and a "Restart" button to the example YAML, so you can reset both the radar module and the ESP32 board too.
- Removed the obsolete rmt_channel from the example YAML LED configuration, so it will compile with the latest version of ESPHome

https://github.com/koszta/MR60BHA2_ESPHome_external_components/blob/main/example/mr60bha2.yaml

limengdu, 2025-04-15:
https://github.com/Seeed-Studio/wiki-documents/pull/2432/files
Updated a version of the radar firmware, please give it a try.

jhansche, 2025-04-16:
@limengdu is there a way to update the firmware without that [custom program](https://wiki.seeedstudio.com/getting_started_with_mr60bha2_mmwave_kit/#module-firmware-upgrade) on the MCU, or without a windows OS to run the OTA tool?

limengdu, 2025-04-16:
Currently, the OTA software provided by chip manufacturers is only available in Windows version, with no other alternatives

closed 2025-04-23 by limengdu

GuyFromMars365, 2025-04-24:
@limengdu why was this ticket marked closed, as it doesn't seem to resolve the problem. Also, can you address @koszta fork, and if that's a better solution, perhaps @koszta could submit a PR? ...I have also noticed inconsistent, unreliable reading, despite updating radar firmware to 1.6.5. Target number is typically inaccurate. I have this deployed in 2 different rooms, just above TV with similar results. Full disclosure, I'm using @koszta forked modification.