limengdu/MR60BHA2_ESPHome_external_components #14: unable to compile firmware with example YAML on version ESPHome 2025.2.2

Kpapadak1, 2025-03-20:
I am attempting to update the firmware of the XI

I used the example in the wii article: https://wiki.seeedstudio.com/ha_with_mr60bha2/

I also tried the example in this repo. I get the same error. 

```
INFO ESPHome 2025.2.2
INFO Reading configuration /config/esphome/mmwave-baby-monitor.yaml...
INFO Updating https://github.com/limengdu/MR60BHA2_ESPHome_external_components@main
Failed config

light.esp32_rmt_led_strip: [source /config/esphome/mmwave-baby-monitor.yaml:59]
  platform: esp32_rmt_led_strip
  id: led_ring
  name: Seeed MR60BHA2 RGB Light
  pin: GPIO1
  num_leds: 1
  
  This feature is not available for the IDF framework version 5.
  rmt_channel: 0
  rgb_order: GRB
  chipset: ws2812
```

Kpapadak1, 2025-03-20:
I commented out the following section and it compiles now but of course now missing LED control

`
#light:
#  - platform: esp32_rmt_led_strip
#    id: led_ring
#    name: "Seeed MR60BHA2 RGB Light"
#    pin: GPIO1
#    num_leds: 1
#    rmt_channel: 0
#    rgb_order: GRB
#    chipset: ws2812
`

koszta, 2025-03-20:
it's enough to delete or comment out

```yaml
rmt_channel: 0
```

closed 2025-03-24 by limengdu