limengdu/MR60BHA2_ESPHome_external_components #15: Distance doesn´t work in the recent Firmware

BangerTech, 2025-04-08:
Just installed the 3.5 release and distance only showss. 0,00 cm

limengdu, 2025-04-15:
Can you provide more information about the test environment for us to evaluate? For example, how is the radar installed? The position of the human body during ranging? Is the human body moving or stationary during ranging? The radar's log information in ESPHome?

limengdu, 2025-04-15:
https://github.com/Seeed-Studio/wiki-documents/pull/2432/files
Updated a version of the radar firmware, please give it a try.

wigledav, 2025-04-15:
Hi limengdu - I have the same issue of missing the distance on 3.5.
Are there some more detailed instructions for using the firmware flash tools please? I have been able to load various yaml configurations from ESPHome on Home Assistant but I am not clear on how to reflash the core system firmware.
Thanks

pabian84, 2025-04-19:
> Hi limengdu - I have the same issue of missing the distance on 3.5. I have followed through the upgrade notes and the arduino install and have managed to reflash the firmware to the latest version and my module is now displaying DISTANCE as well as its other parameters. Looking good! Thanks

Hello. I’m experiencing the same problem as you. I’m familiar with ESPHome and YAML, but I have no knowledge of Arduino at all. Could you share any information that would help me follow your solution?

wigledav, 2025-04-19:
Sure. I made some reasonable notes on how to do this as it took quite a few hours to explore. I'm out tonight but will send them over tmrw.
This document is fairly key:
https://wiki.seeedstudio.com/xiao_esp32c6_getting_started/

It explains the Arduino install and adding xiao esp32c6 board manager and getting the esp32c6 board into pass through mode

Then you need the actual pass through code to cut and paste into the Arduino ide:

https://github.com/Seeed-Studio/wiki-documents/pull/2432/files

In this long page you will see code that looks like this:
[cid:img-d5fe52de-7681-4e1a-a64c-165b6f91a27c]
This is the pass through code. Copy and paste it into Arduino and upload to esp board.

Then you need the ota firmware update program:

https://github.com/Seeed-Studio/wiki-documents/pull/2432/files

See line 570 or 572 and download the ota program.

Also download the latest 1.6.5 firmware from this page.

https://files.seeedstudio.com/wiki/mmwave-for-xiao/mr60/firmware/MR60BHA2_eeprom_1.6.5.bin

Run the ota program and upload the new firmware as per detailed instructions in the second half of this :

https://wiki.seeedstudio.com/getting_started_with_mr60bha2_mmwave_kit/#module-firmware-upgrade

Hopefully that will get you going with 1.6.5. I've done it largely from memory so hope correct.

I can send you my yaml tmrw. See how you get on.

Dave

Dave Wigley

rickt123, 2025-04-19:
@wigledav is there anyway to update the firmware?  I think I have a defective unit, but trying to update the firmware based on communications with Seeed Tech Support.  However I just can't see to get the OTA tool run correctly on Win10/11.  Whenever I request the update, I get the first prompt, "Whether to upgrade the MCU firmware."  But the second prompt never appears. Thanks for any insigths.

pabian84, 2025-04-20:
@wigledav  Thank you for your reply. I’m trying to follow your instructions, but I have a few questions:

1. I don’t understand the step that says “Copy and paste it into Arduino and upload to the ESP board.” Where exactly should I paste the code?
2. What is the difference between the .bin file built from the YAML and the .bin file for firmware version 1.6.5?
3. Should I first update the board with the 1.6.5 firmware and then build and upload your YAML afterward?
4. 
It was hard to follow along with just Seeed’s wiki page, so thank you for providing the guidance.

wigledav, 2025-04-20:
I have had a go at outlining my 'start from scratch' instructions. See attached PDF. I have only had the unit a few days and am NOT an ESP or radar expert but have successfully flashed both the low-level radar firmware and the higher-level YAML code multiple times. I hope my PDF helps. It is undoubtedly a fiddly process.
Dave

Note that I see quite difference performance between radar firmware 1.6.4 and 1.6.5 with 1.6.4 offering longer-range presence detection but flakier heart rate performance.

[MR60HBA2 Full Flash Instructions - DRAFT.pdf](https://github.com/user-attachments/files/19824997/MR60HBA2.Full.Flash.Instructions.-.DRAFT.pdf)

pabian84, 2025-04-20:
@wigledav  Thank you very much for even preparing the PDF file!

After many attempts, it looks like I finally succeeded in updating one unit—the radar module and ESP32 board are now running the latest firmware. However, on the second unit I accidentally loaded version 1.6.4 and haven’t been able to recover it: the OTA Tool receives no data after connecting, doesn’t respond to “REQUEST UPDATE,” and even flashing just the YAML fails, so it seems something may be bricked.

I bought this product because the ads said it could be used very easily without any coding, so I’m really frustrated that—even with all this help—it’s still so difficult and isn’t working properly.

[image]

Thanks to you, I now have at least one successful experience. I might even purchase additional units if needed. Thank you.

pabian84, 2025-04-21:
@wigledav 
I flashed the Arduino serial comms code onto the ESP32. However, unlike the first time, no logs appear in the OTA Tool, and pressing “REQUEST UPDATE” elicits no response. I’ve tried resetting the board and holding down the boot button during the process, but it still won’t work.
And, most importantly, when I tested the unit that was successfully updated, I discovered that the human detection and heart‑rate detection features cannot be disabled.
I greatly appreciate your kind assistance, yet I’m disappointed that I haven’t succeeded. Moreover, using a product advertised as “no‑code” in such a stressful way has been exhausting, so I think I need to take a short break. Next week, I will follow your suggestion to reboot the device and try again.

limengdu, 2025-04-21:
Sorry, we apologize for the inconvenience caused to everyone when using this radar. As the function and algorithm of the radar are constantly updated and iterated, we will also release new firmware versions to everyone. We try to ensure that each firmware version can function properly to avoid users needing to frequently update the product's firmware. However, since the radar itself does not have network functionality and firmware reception depends on commands, the firmware burning process for the radar is quite cumbersome. We will strive to improve the firmware update experience in the future.

OliverBuschmann, 2025-05-21:
Hi @limengdu when can we expect a new version that fixes this distance issue and optimises the sensors please? Aka 3.6 release? I really can not accept to buy a factory set up sensor that doesnt really work with all its features. I have no ability to use the IDE 's etc thats why I bought the ready to go kit. Can you give us an update to when?

limengdu, 2025-05-23:
Our latest firmware V1.6.10 has been released, welcome to update and use it:   

- **MR60BHA2 Firmware v1.6.10**: [MR60BHA2_eeprom_1.6.10.bin](https://files.seeedstudio.com/wiki/mmwave-for-xiao/mr60/firmware/MR60BHA2_eeprom_1.6.10.bin)

gretel, 2025-05-23:
> Our latest firmware V1.6.10 has been released

@limengdu thank you! can you provide a change log please?

wigledav, 2025-05-23:
I have successfully downloaded this firmware 1.6.10 and the device is now identifying more distant objects. However, my initial feedback is that it is dropping presence very quickly such that if you stand still (at a distance outside the heartrate zone) it rapidly loses the subject. And if you sit still in front of the sensor to measure heartrate it also drops out. See attached over a few minutes of data capture sitting in front of the sensor.

[image]

limengdu, 2025-05-26:
@wigledav Yes, we will continue to fix these issues. Currently, it seems that there is interference between various functions, and we need more time to further optimize the algorithm

ifajardoj, 2025-06-24:
> I have successfully downloaded this firmware 1.6.10 and the device is now identifying more distant objects up to 500cms. However, my initial feedback is that it is dropping presence very quickly such that if you stand still (at a distance outside the heartrate zone) it rapidly loses the subject. And if you sit still in front of the sensor to measure heartrate it also drops out. See attached over a few minutes of data capture sitting in front of the sensor.
> 

Same behavior here with firmware 1.6.10, above 1.5/1.8 m, presence, distance, etc dropping in 5-10 seconds depending on the distance to the person.

limengdu, 2026-10-15:
A fix for distance on recent radar firmware (1.6.10 / 1.6.12) is in PR #29. Could you test it with `github://limengdu/MR60BHA2_ESPHome_external_components@limengdufix-distance-firmware-945b` (`refresh: 0s`) and check that distance no longer stays at 0.00 cm while standing or sitting? https://github.com/limengdu/MR60BHA2_ESPHome_external_components/pull/29