limengdu/MR60BHA2_ESPHome_external_components #17: MR60HBA2 and MR60FPA2 - No (useful) distance presence sensing - new firmware for both please

wigledav, 2025-04-22:
These modules are sold as offering presence sensing up to 6 metres from the unit:

"The MR60BHA2 60GHz mmWave Sensor Module excels in detecting breathing and heartbeat with pre-set algorithms, and also includes human static presence functionality.

Human Static Presence: Measures effectively up to **6 meters**.
Breathing & Heartbeat Detection: Accurate for a single person at 1.5 meters with the device stationary."

However, neither module will provide any reliable presence sensing outside 1.8 metres as distance tracking freezes or does not re-enable when a user moves into the 6 meter zone, only restarting once inside about 1.5 meters.

Please provide a new version of firmware that delivers a reliable presence sense X/Y or distance measure outside of the heartrate sense zone. From using the Seeed GUI tool it is clear that he data is often available but is not reliably delivered by the firmware. Perhaps this can be a separate set of measures from the 'close' zone.

Thanks

Dave

limengdu, 2025-05-06:
Currently, it has been discovered that the data reporting of the radar at around 1.8 meters gets stuck. Engineers are working hard to fix it, please wait a moment.

wigledav, 2025-05-06:
Thank you for acknowledging the issue I reported. I look forward to a firmware fix shortly.

Happy to test some beta code for you.

Dave

binary100100, 2025-05-30:
I bought three of these from Seeedstudio with the intention of using it as a presence sensor. 
However, they randomly seem to get stuck at "Detected".

In this image... I am in the office at this very moment. 
[image]

But I'm absolutely not in my bedroom with a heartrate of 66bpm.

[image]

I've come to the conclusion that even with the newest firmware (3.5), I can't use these for their expected purpose.
Hoping that changes soon though.

But on a slightly different topic... what is "Target Number" all about?
Is that how many subjects are detected?
If so, I have ghosts!

[image]

gretel, 2025-06-08:
same for me.

Motya-HA, 2025-07-24:
Just bought two of them. After half day of hassling and updating firmware I am getting this

[image]

Light sensor is working, but anything related to mmWave is not
I was really counting on the sensor, please help

> You should try updating the radar's firmware, not the ESPHome firmware

How?

Motya-HA, 2025-07-24:
I've tried to add it with ESPHome (add device)

`# template from https://github.com/limengdu/MR60BHA2_ESPHome_external_components

substitutions:
  name: "sleep-test-mr60bha2-kit"
  friendly_name: "SSleep Test"

esphome:
  name: "${name}"
  friendly_name: "${friendly_name}"
  name_add_mac_suffix: true
  project:
    name: "seeedstudio.mr60bha2_kit"
    version: "2.0"
  platformio_options:
    board_upload.maximum_size: 4194304
  min_version: "2024.3.2" # Fix logger compile error on ESP32-C6 esphome#6323

esp32:
  board: esp32-c6-devkitc-1
  variant: esp32c6
  flash_size: 4MB # upload.flash_size
  framework:
    type: esp-idf

external_components:
  - source:
      type: git
      url: https://github.com/limengdu/MR60BHA2_ESPHome_external_components
      ref: main
    components: [ seeed_mr60bha2 ]
    refresh: 0s

/# Enable logging
logger:
  hardware_uart: USB_SERIAL_JTAG
  level: DEBUG

/# Enable Home Assistant API
api:
  encryption:
    key: "xxxxxxxxxxxxxxxxxxxx"

ota:
  - platform: esphome
    password: "xxxxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

 / # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Sleep-Test-Esp Fallback Hotspot"
    password: "ITBxq8tMvHLB"

captive_portal:

/# For XIAO ESP32C6 Onboard LED
/# light:
/#   - platform: status_led
/#     name: "Switch state"
/#     pin: GPIO15

/# light:
/#   - platform: esp32_rmt_led_strip
/#     id: led_ring
/#     name: "Seeed MR60BHA2 RGB Light"
/#     pin: GPIO1
/#     num_leds: 1
/#     rmt_channel: 0
/#     rgb_order: GRB
/#     chipset: ws2812

i2c:
  sda: GPIO22
  scl: GPIO23
  scan: true
  id: bus_a

uart:
  id: uart_bus
  baud_rate: 115200
  rx_pin: 17
  tx_pin: 16
  parity: NONE
  stop_bits: 1

seeed_mr60bha2:
  id: my_seeed_mr60bha2

sensor:
  - platform: bh1750
    name: "Seeed MR60BHA2 Illuminance"
    address: 0x23
    update_interval: 1s
  - platform: seeed_mr60bha2
    breath_rate:
      name: "Real-time respiratory rate"
    heart_rate:
      name: "Real-time heart rate"
    distance:
      name: "Distance to detection object"`

I had to comment block

```
 light:
 - platform: esp32_rmt_led_strip
    id: led_ring
    name: "Seeed MR60BHA2 RGB Light"
    pin: GPIO1
    num_leds: 1
    rmt_channel: 0
    rgb_order: GRB
    chipset: ws2812

```
becuase it was flagging 
In any event, it did not work

Here is log
[logs_sleep-test-esp_compile.txt](https://github.com/user-attachments/files/21420424/logs_sleep-test-esp_compile.txt)

limengdu, 2025-07-25:
If you are using the latest code, you should be using version 3.5 of YAML. 
https://github.com/limengdu/MR60BHA2_ESPHome_external_components/blob/ de92a962e2bfa5342225f83a48b1f422b984318a/example/mr60bha2.yaml#L11

Motya-HA, 2025-07-25:
The screen shot is from flashed firmware in ESPHome Web and .bin file: https://github.com/limengdu/MR60BHA2_ESPHome_external_components/releases/download/V3.5/seeedstudio-mr60bha2-kit-esp32c6.factory.bin, which is in MR60BHA2 mmWave Kit Firmware V3.5 directory.

I could not install with ESPHome and yaml code, which i took from your read.me file (which you probably need to update)
But for now. let's forget about ESPHome. Why flashing from Web Tool and ESPHome web with firmware file with the *.factory.bin suffix gives me "unknown" entities for presence

limengdu, 2025-07-25:
The current findings indicate that the known issue where entities display "Unknown" may occur due to insufficient power supply to the radar. Specifically, if the instantaneous current during startup is inadequate, the radar may fail to initialize and operate properly. No other anomalies have been reported besides this. It is recommended to use a 5V 2A power adapter or a compliant adapter that supports higher current capacity.

Motya-HA, 2025-07-25:
Ok.
I followed your directions. I flashed sensor again with Web tool (BTW, it can be flashed only if connected to USB-C on the XIAO ESP32C6 itself (so you need to take a sensor for the plastic mount).

After using 
[seeedstudio-mr60fda2-kit-esp32c6.factory.bin](https://github.com/limengdu/MR60FDA2_ESPHome_external_components/releases/download/V2.1/seeedstudio-mr60fda2-kit-esp32c6.factory.bin) as advised, this is what I get. Sometimes it stuck in "detected" without reason.
 

[image]

I've created new device in ESPHome with your yamal code (rev 3.5), however, I see the following:

[image]

Again, Illuminance and LED is working, presence not.

Another observation, I can't make with device to be onlne with WiFi, even it is connected to my WiFi.

Attached yamal code:
`substitutions:
  name: "seeedstudio-mr60bha2-kit"
  friendly_name: "Seeed Studio MR60BHA2 Kit"

esphome:
  name: "${name}"
  friendly_name: "${friendly_name}"
  name_add_mac_suffix: true
  project:
    name: "seeedstudio.mr60bha2_kit"
    version: "3.5"
  platformio_options:
    board_upload.maximum_size: 4194304
  min_version: "2024.3.2" # Fix logger compile error on ESP32-C6 esphome#6323

esp32:
  board: esp32-c6-devkitc-1
  variant: esp32c6
  flash_size: 4MB # upload.flash_size
  framework:
    type: esp-idf

external_components:
  - source:
      type: git
      url: https://github.com/limengdu/MR60BHA2_ESPHome_external_components
      ref: main
    components: [ seeed_mr60bha2 ]
    refresh: 0s

# Enable logging
logger:
  hardware_uart: USB_SERIAL_JTAG
  level: DEBUG

api:
  encryption:
    key: "xxxxxxxxxxxxxxxxxxxxx"

ota:
  - platform: esphome
    password: "xxxxxxxxxx"

wifi:
  # ssid: !secret wifi_ssid
  # password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "seeedstudio-mr60bha2"

captive_portal:

# For XIAO ESP32C6 Onboard LED
# light:
#   - platform: status_led
#     name: "Switch state"
#     pin: GPIO15

light:
  - platform: esp32_rmt_led_strip
    id: led_ring
    name: "Seeed MR60BHA2 RGB Light"
    pin: GPIO1
    num_leds: 1
    rgb_order: GRB
    chipset: ws2812

i2c:
  sda: GPIO22
  scl: GPIO23
  scan: true
  id: bus_a

uart:
  id: uart_bus
  baud_rate: 115200
  rx_pin: 17
  tx_pin: 16
  parity: NONE
  stop_bits: 1

seeed_mr60bha2:
  id: my_seeed_mr60bha2

binary_sensor:
  - platform: seeed_mr60bha2
    has_target:
      name: "Person Information"

sensor:
  - platform: bh1750
    name: "Seeed MR60BHA2 Illuminance"
    address: 0x23
    update_interval: 1s
  - platform: seeed_mr60bha2
    breath_rate:
      name: "Real-time respiratory rate"
    heart_rate:
      name: "Real-time heart rate"
    distance:
      name: "Distance to detection object"
    num_targets:
      name: "Target Number"
    `
Ohh, I forgot to mention I am using Apple MacBook Pro power supply (with USB-C connector).
If this is not enough current, next step will be Tesla charging station (before going to nuclear Power plant

limengdu, 2025-07-28:
Being stuck in detected state is a known issue with the current radar firmware. We are updating the software algorithm, please wait.

limengdu, 2025-07-28:
我才看到您问的是MR60FDA2这款雷达，你可以在对应的仓库提交问题吗？https://github.com/limengdu/MR60FDA2_ESPHome_external_components

limengdu, 2025-07-28:
The example code you gave is still for MR60BHA2, if the product you are using is MR60FDA2, then the “Unknown” value you got is normal, please check what equipment you are using.

Motya-HA, 2025-07-29:
> folks mainly looking for distance (and position) are better off using the much cheaper LD2450 sensor. recent release of `esphome` [fixes quite some issues](https://github.com/esphome/esphome/pull/9849).

I am looking for the static detection. 70G supposed to be better (for detecting sleeping person).
In any event, I bought this sensor (2 of them) and I want them work

binary100100, 2025-07-29:
> folks mainly looking for distance (and position) are better off using the much cheaper LD2450 sensor. recent release of `esphome` [fixes quite some issues](https://github.com/esphome/esphome/pull/9849).

> 
> I am looking for the static detection. 70G supposed to be better (for detecting sleeping person). In any event, I bought this sensor (2 of them) and I want them work

I gave up on these.
They're not reliable. The project kinda looks abandoned.
Too many people are complaining about the same issues and nothing has been corrected or hardly even addressed. And I get it... the developers involved have lives and don't profit much (if at all) from it. But I won't be buying more kits from Seeedstudio and don't recommend anyone else doing so either.
Instead of waiting for reliable products to work as advertised I bought three Aqara FP2 devices. They work flawlessly! Absolutely no complaints. 
But I'd still like my three sensors from Seeedstudio to work.

binary100100, 2025-07-29:
> Response from Seeed technical team:
> 
> Hi,
> 
> regarding the issue you mentioned, we will test it for you as soon as possible.
> 
> Best Regards!
> 
> Seeed Technical Support Team
> 
> For some reason I'm not holding my breath :-)

People have been complaining about this issue for over three months.
There hasn't been a firmware update (after 3.5) in almost six months.
If a project has gone six months without any updates... I consider that mostly dead.
The fact that Seeedstudio still sells something advertising features that aren't available and offers ZERO support to fix in this length of time is unfortunate. But... you get what you pay for. 
I've bought quite a few things from them. Can't say that I had good or bad experiences. I bought my HomeAssistant Yellow POE from them, my three HomeAssistant Voice PE units and my three heartbeat/presence sensors. The only issue I've had was these stupid unreliable sensors. The instructions to update are too confusing because there are too many variants and too many methods and shit for guides. I spent two days trying to use the instructions that were telling me to download Arduino and I couldn't ever get it to work. Then tried the webinstaller which was a pain in the ass to find in all of the instructions floating around for the different versions but that installs 3.5 which we all know is far from perfect. 
So... yeah... I've given up on it a while ago and went with the Aqara FP2 presence sensors. They don't think they do the heartbeat AND the presence sensing but they're much more reliable in my opinion. A bit pricer but definitely worth it though.

wigledav, 2025-07-29:
Same experiences here. Waste of money and will avoid Seeed in future.
I ended up buying a couple of Tuya 24ghz sensors with WiFi and they work flawlessly for presence detection. At the near top setting I only need to twitch an arm and presence is continued. So with a one minute timeout you get a practical presence sensor for about one quarter of the cost and one hundredth of the time effort required to make it work.

Dave Wigley

limengdu, 2025-07-29:
@binary100100 The radar product line has not been "abandoned" as claimed. In fact, our software team and algorithm team have been continuously working on firmware updates. Millimeter-wave radar technology currently represents a relatively high technical threshold, and we often lack extensive experience in dealing with certain technical challenges, which results in software iteration cycles typically taking several months. On behalf of the entire team, I sincerely apologize for this delay. 

I'd like to clarify further that we won't be frequently updating the ESPHome firmware in this repository because its primary purpose is to parse and display the radar-reported data in Home Assistant. In other words, if the radar's functionality is stable and its communication protocol remains unchanged, there's generally no need to update the ESPHome firmware. Our software team's main focus remains on improving the radar's native firmware, particularly its performance and algorithmic aspects.
For example, issues like "ghosting" (false readings when empty) and breathing/heart rate detection in unoccupied spaces discussed above are essentially problems with the radar's core detection capabilities. This requires us to reproduce various scenarios and refine algorithmic details to enhance the radar's firmware. 

This year alone, we've released three firmware versions, and we're planning another update by late August to early September specifically targeting the interference issues between human presence detection and simultaneous breathing/heart rate monitoring. We truly appreciate your patience and ongoing feedback.

limengdu, 2025-07-29:
Thank you for your understanding. We'll do our best to release a firmware version that can first ensure stable output results for these functionalities in one scenario. I will also submit your suggestions and feedback to the radar development team.

gretel, 2025-07-29:
@limengdu launching a new device without fixing issues on the current one is why i am diregarding the `Seeed` brand as a whole. lots of words can't make up for technical debts.

limengdu, 2025-07-31:
We updated the Wiki in May with firmware version 1.6.10, and in April with 1.6.5. As I explained in the previous post, the update you are referring to is an update of the ESPHome code for this repository, which does not need to be updated if the protocol remains unchanged. What we are working on is a firmware update for the radar, not an ESPHome update.

1.6.10 Added region settings, but currently unable to save settings to flash, settings are cleared when power is lost
1.6.9 Position stuck at 1.8 meters bug (no data collection function)
1.6.8 Added data collection function
1.6.7 Optimized breathing, fixed issue where breath-holding values need to drop to 0
1.6.6 Optimized breathing, fixed issue where breath-holding values don't decrease

binary100100, 2025-07-31:
> We updated the Wiki in May with firmware version 1.6.10, and in April with 1.6.5. As I explained in the previous post, the update you are referring to is an update of the ESPHome code for this repository, which does not need to be updated if the protocol remains unchanged. What we are working on is a firmware update for the radar, not an ESPHome update.
> 
> 1.6.10 Added region settings, but currently unable to save settings to flash, settings are cleared when power is lost 1.6.9 Position stuck at 1.8 meters bug (no data collection function) 1.6.8 Added data collection function 1.6.7 Optimized breathing, fixed issue where breath-holding values need to drop to 0 1.6.6 Optimized breathing, fixed issue where breath-holding values don't decrease
> 
> If you are updating all the way down, you should see that the radar bugs have been there but the features are slowly being added. From the beginning when it was first released with only human presence detection and respiratory heart rate detection, to the addition of ambient headcount, and then the addition of distance detection. To We have been maintaining the firmware of the radar so that the chip inside can play as many roles as possible in a variety of scenarios. There is no such thing as stopping maintenance and updating. the ESPHome code is the finalised version and does not need to be updated frequently, we update the radar firmware on the [Wiki](https://wiki.seeedstudio.com/getting_started_with_mr60bha2_mmwave_kit/#module-firmware-upgrade).

See?
That's what I'm talking about. 
It's too complicated with all is the firmware versions, methods of flashing and hardware variations.
You say that the ESPHome is finalized and doesn't need  updating when clearly it doesn't fully work! This is what many of us have been complaining about.
Nevermind. I completely give up on these.
I'm never buying shit from Seeedstudio again. Support absolutely sucks and I shouldn't have spent this $100 on this junk in the first place.

limengdu, 2025-08-01:
The current issue is caused by incomplete radar firmware and has no relation to ESPHome firmware. Updating the ESPHome firmware will not resolve the problem. Please wait for the new radar firmware expected in late August or early September to try again. Thank you for your support, and we sincerely apologize for any inconvenience caused.

gretel, 2025-08-03:
> Please wait for the new radar firmware expected in late August or early September to try again. Thank you for your support, and we sincerely apologize for any inconvenience caused.

the issue has been know since [may](https://github.com/limengdu/MR60BHA2_ESPHome_external_components/issues/17#issuecomment-2853124760) - that's going to be six months at least. worse, it's hard to believe another version will actually assure quality.

GuyFromMars365, 2025-09-01:
Let me re-emphasize a few points to help everyone. The RADAR firmware updates and the ESPHome YAML updates are 2 different pieces of software. The ESPHome YAML can be updated OTA, while the RADAR board can ONLY be updated via USB inside the case, with specialized software that SEEED provides. You can customize the ESPHome software, you can NOT customize the RADAR firmware. SEEED only allows RADAR customization if your doing a business application. I tried to update their Wiki and Github with this clarification, but despite multiple email complaints, they still haven't clarified this on their website. 

[image]

@limengdu I hope SEEED can learn from their mistakes on this project.  ...When it was first released, the SEEED website did NOT mention this as a Early Build/BETA (that was added after I purchased Dec, '24).  It was marketed as a customizable radar sensor with software ready to work with Home Assistant. In reality, it's a beta non-customizable radar sensor on a customizable ESP32 board that you could tweak/ad settings to. That might still be ok for some people if that's what was clearly advertised, but it isn't. Please have your company clarify/fix this mistake on their website and be clear with products in the future.

Like others, I will be thinking twice before purchasing another SEEED product. You'll always have people that don't read docs and complain (sorry, sounds like some in this thread), but you can minimize that by clear docs, so you set proper expectations. False advertising with unclear documentation creates the kind of problem we see now.

limengdu, 2025-09-19:
@rickt123 For firmware flashing issues, please contact Seeed's technical support team (techsupport@seeed.cc) directly. They will guide you through the process and check if your device has any problems.

binary100100, 2025-09-19:
I just tried this as well.
Now I'm seeing just a blank Seeed Studio OTA screen.

[image]

Green icon in the lower left. Radar: NULL Status: COM3 opened successfully!
Nothing happens when I select any of the buttons.

Still comes up on Arduino though.

[image]

I connected one of my other identical devices:

[image]
Doesn't seem to have any issues. So it's not the Seeed OTA program.

I've done this a few times before. But this isn't going well.

limengdu, 2025-09-19:
The firmware was released after undergoing two rounds of testing. I will record a complete operation video on Monday to help everyone flash the firmware.

binary100100, 2025-09-19:
> The firmware was released after undergoing two rounds of testing. I will record a complete operation video on Monday to help everyone flash the firmware.

So the people that tested didn't use the same instructions that rickt123 and I followed apparently. It's weird that we both ended up with the same results and now with useless devices. Perhaps it's because some people have different boards? I'm using the XAIO ESP32C6. I dunno what happened. All I can say is that after flashing with the Seeed OTA app, it didn't display a radar information or anything. After the process, I unplugged it and flashed the Seeedstudio firmware to set it back up with HomeAssistant and no information was being displayed. I tried flashing it again and now nothing is showing up on Seeed Studio OTA. Nothing at all. It says it's connected, but nothing is happening. I use the Arduino software to flash the heartbeat and breathing to it but still nothing (except for illuminance) in HomeAssistant.

rickt123, 2025-09-19:
> The firmware was released after undergoing two rounds of testing. I will record a complete operation video on Monday to help everyone flash the firmware.

@limengdu I really hope you follow the steps you pointed all of us to follow

binary100100, 2025-09-19:
I bought a UART 3.3 to try to fix this thing.
After flashing the MR60BHA2_eeprom_1.6.12.bin file with esp-idf-v5.1.1 all it's doing is causing my computer to beep like a device is connected and then disconnects. Now I really can't do anything with this.
Gonna try esp-idf-v5.1.1 again (if I can get bootloader at least) and try the old 1.6.10.bin and hopefully I can salvage this damn thing.

[image]

Pr0mises, 2025-09-19:
> The firmware was released after undergoing two rounds of testing. I will record a complete operation video on Monday to help everyone flash the firmware.

Just to let you know, I was able to flash 1.6.12 with esp32c6 but the radar didn't get better while sitting within a 1.5m range. I have to move all the time

rickt123, 2025-09-19:
Did you follow a different set of steps?  Mine seemed to have been successfully as well but killed the radar.

binary100100, 2025-09-19:
> I used those instructions: https://wiki.seeedstudio.com/getting_started_with_mr60bha2_mmwave_kit/#module-firmware-upgrade
> 
> Plugged both together, plugged the usb c into the esp32c6 and followed the instructions

That's the exact set of instructions that I've been using.
Step by step. No rushing. Followed every word.

Pr0mises, 2025-09-19:
Did you use the arduino code beforehand and got some garbage output? Had version 1.6.10 before the new version installed

rickt123, 2025-09-19:
Thanks glad it was successful for you.

https://wiki.seeedstudio.com/getting_started_with_mr60bha2_mmwave_kit/#module-firmware-upgrade

rickt123, 2025-09-19:
Exactly.

binary100100, 2025-09-20:
> [@binary100100](https://github.com/binary100100) [@rickt123](https://github.com/rickt123) Do you have a UART-to-USB converter module? If so, you can try connecting the radar directly to your computer via the UART-to-USB module, then follow the steps outlined in the firmware upgrade tool. The upgrade process can be performed without using XIAO. Remember to upgrade the software. If the radar does not print “C,” it means the radar has not entered upgrade mode, and you cannot proceed with the subsequent steps.

I do have one. Bought it today trying to see if I can fix this. Nothing shows up with OTA tool. I was able to flash the firmware espflash.py tool but no difference. Everything on the OTA tool is blank and no matter what I do everything shows "unavailable" in HomeAssistant.

rickt123, 2025-09-20:
I unfortunately do not have one.

ifajardoj, 2025-09-20:
> [@binary100100](https://github.com/binary100100) [@rickt123](https://github.com/rickt123) Do you have a UART-to-USB converter module? If so, you can try connecting the radar directly to your [...]

Can you give more info on how to do this? How to make the connections?

binary100100, 2025-09-20:
> Can you give more info on how to do this? How to make the connections?

https://www.espboards.dev/esp32/xiao-esp32c6/

[image]

ifajardoj, 2025-09-20:
> https://www.espboards.dev/esp32/xiao-esp32c6/
>
> [image]

I try this way but OTA software still blank, how can I put the radar in upgrade mode?

rickt123, 2025-09-20:
@binary100100 would you mind sharing with converter module you purchased?

binary100100, 2025-09-20:
> I try this way but OTA software still blank, how can I put the radar in upgrade mode?

Yeah. Me too.

limengdu, 2025-09-22:
Hi @gretel @Pr0mises @binary100100 @ifajardoj @rickt123 , I've invited you to my private repository [temp-contact](https://github.com/limengdu/temp-contact). I hope to connect with you soon. Please check if you've received the invitation.

ifajardoj, 2025-09-22:
> Hi [@gretel](https://github.com/gretel) [@Pr0mises](https://github.com/Pr0mises) [@binary100100](https://github.com/binary100100) [@ifajardoj](https://github.com/ifajardoj) [...]

Done.
Thanks.

Pr0mises, 2025-09-22:
> Hi @gretel @Pr0mises @binary100100 @ifajardoj @rickt123 , I've invited you to my private repository [temp-contact](https://github.com/limengdu/temp-contact/issues/1). I hope to connect with you soon. Please check if you've received the invitation.

Done thanks

limengdu, 2025-09-25:
https://www.youtube.com/watch?v=uic8__FFzos
The firmware update video has been uploaded to YouTube. Starting at 1 minute and 42 seconds, we briefly demonstrated the testing results for firmware version 1.6.12.

Pr0mises, 2025-09-25:
> > https://www.youtube.com/watch?v=uic8__FFzos The firmware update video has been uploaded to YouTube. Starting at 1 minute and 42 seconds, we briefly demonstrated the testing results for firmware version 1.6.12.
> 
> So guess what?
> It's not the esp32c6 that's bad.
> SOMEHOW, I borked the 60GHz mmWave Sensor Module part.
> 
> I tried all three of these brand new esp32c6 and NONE of them would work. Very odd!
> So I took another one of my three previously flashed and installed sensors (that I had placed in the "Living Room") and put in the old esp32c6 that should have the .12 firmware on it. (Unless I re-flashed the .10 over it trying to fix it? I dunno.) Call that one "Office" because that's where I had it.
> And it started working! I put the "Living Room" esp32c6 in the 60GHz mmWave Sensor Module that was in the office and everything showed "unavailable" in Home Assistant but Office was working fine. I swapped them back around again (Office back in it's original 60GHz mmWave Sensor Module and 3D printed housing) and it wasn't working. So... it appears to be a problem with the 60GHz mmWave Sensor Module itself. Somehow, that got screwed up. 
> I took it apart and I don't have any way to tell what or if it's damaged somehow. Guess I'll order a new one. :(

Accept his private repo and go into the issues

rickt123, 2025-09-25:
> https://www.youtube.com/watch?v=uic8__FFzos The firmware update video has been uploaded to YouTube. Starting at 1 minute and 42 seconds, we briefly demonstrated the testing results for firmware version 1.6.12.

There are the exact same instructions I followed.. although I think I have a different version of the firmware tool

[image]

pilonull, 2025-11-02:
@limengdu  are you the one working on firmware updates for the radar module itself? 

Asking because https://wiki.seeedstudio.com/getting_started_with_mr60bha2_mmwave_kit/#module-firmware-upgrade

says:

> A firmware update to improve breathing and heart rate accuracy will be released around month-end.

But no indication of what month that's referring to.

My issue is that, even on current latest firmware, 1.6.12, is that with esphome the distance to detection object, Person information, heart rate, respiratory rate, and target number values all seem to "hang"

Like, Target number is always 1 even if there is nobody in front of the sensor. Sometimes it will go up to 2, then go back down to 1. And I guess since target number is at 1, it always detects a person even though there is no person.

Also, when it DOES detect a heart rate or respiratory rate, it'll get a number that seems valid, then the person will move away from the sensor, and that number remains there until a person comes in front of the sensor again. Sometimes the number will hang even if the person remains in front of the sensor. 

It's all very inconsistent and unreliable, unfortunately. 

I was really hoping to use this on my newborn's crib to monitor them but I just can't trust it in it's current state.

gretel, 2025-11-02:
check out [t-radar](https://lilygo.cc/products/t-radar) based on `A121` with plenty of [documentation](https://docs.acconeer.com/en/latest/detectors/a121/presence_detector.html) - regards

limengdu, 2026-10-15:
PR #29 now exposes the far-zone X/Y coordinates and point cloud the Seeed GUI uses, adds optional `target_x` / `target_y` sensors, and treats presence as true if any of four signals detects a target. Please test with `github://limengdu/MR60BHA2_ESPHome_external_components@limengdufix-distance-firmware-945b` (`refresh: 0s`): walk from ~1.5 m out to 3–6 m and check that `has_target` stays on and the coordinates follow you. Note: if the radar itself stops reporting around 1.8 m (also visible in the GUI), that's a radar firmware limit ESPHome can't fix. MR60FPA2/FDA2 are handled in the MR60FDA2 repo. https://github.com/limengdu/MR60BHA2_ESPHome_external_components/pull/29