limengdu/MR60BHA2_ESPHome_external_components #18: Person Information never returns to cleared

rickt123, 2025-05-02:
Hello I've installed all of the latest firnware, esphome code and successfully integrated with Home Assistant.  Unfortunately once the device detect a person it never clears it.  I have this device in a room that get's little foot traffic and it (I think) correctly detected when I came into range, however after waiting several hours it's reporting detected and more interesting it's also distance, heart and respiratory rates and targets.  I can reboot it, but again that's only temporary and the next time someone comes into range it will just get stuck there.

Here's the log output
INFO Starting log output from 192.168.xxx.xxx using esphome API
INFO Successfully connected to seeedstudio-mr60bha2-kit @ 192.168.xxx.xxx in 4.116s
INFO Successful handshake with seeedstudio-mr60bha2-kit-b4b994 @ 192.168.xxx.xxx in 0.031s
[14:15:05][I][app:100]: ESPHome version 2025.4.1 compiled on May  2 2025, 14:14:45
[14:15:05][I][app:102]: Project seeedstudio.mr60bha2_kit version 2.0
[14:15:05][C][wifi:600]: WiFi:
[14:15:05][C][wifi:428]:   Local MAC: XX:XX:XX:XX:XX:XX
[14:15:05][C][wifi:433]:   SSID: [redacted]
[14:15:05][C][wifi:436]:   IP Address: 192.168.xxx.xxx
[14:15:05][C][wifi:439]:   BSSID: [redacted]
[14:15:05][C][wifi:441]:   Hostname: 'seeedstudio-mr60bha2-kit-b4b994'
[14:15:05][C][wifi:443]:   Signal strength: -60 dB ▂▄▆█
[14:15:05][C][wifi:447]:   Channel: 11
[14:15:05][C][wifi:448]:   Subnet: 255.255.255.0
[14:15:05][C][wifi:449]:   Gateway: 192.168.xxx.xxx
[14:15:05][C][wifi:450]:   DNS1: 1.1.1.1
[14:15:05][C][wifi:451]:   DNS2: 1.0.0.1
[14:15:05][C][logger:177]: Logger:
[14:15:05][C][logger:178]:   Max Level: DEBUG
[14:15:05][C][i2c.idf:083]: I2C Bus:
[14:15:05][C][i2c.idf:084]:   SDA Pin: GPIO22
[14:15:05][C][i2c.idf:085]:   SCL Pin: GPIO23
[14:15:05][C][i2c.idf:086]:   Frequency: 50000 Hz
[14:15:05][C][i2c.idf:092]:   Recovery: bus successfully recovered
[14:15:05][I][i2c.idf:102]: Results from i2c bus scan:
[14:15:05][I][i2c.idf:108]: Found i2c device at address 0x23
[14:15:05][C][uart.idf:159]: UART Bus 0:
[14:15:05][C][uart.idf:160]:   TX Pin: GPIO16
[14:15:05][C][uart.idf:161]:   RX Pin: GPIO17
[14:15:05][C][uart.idf:163]:   RX Buffer Size: 256
[14:15:05][C][uart.idf:165]:   Baud Rate: 115200 baud
[14:15:05][C][uart.idf:166]:   Data Bits: 8
[14:15:05][C][uart.idf:167]:   Parity: NONE
[14:15:05][C][uart.idf:168]:   Stop bits: 1
[14:15:05][C][esp32_rmt_led_strip:250]: ESP32 RMT LED Strip:
[14:15:05][C][esp32_rmt_led_strip:251]:   Pin: 1
[14:15:05][C][esp32_rmt_led_strip:253]:   RMT Symbols: 96
[14:15:05][C][esp32_rmt_led_strip:281]:   RGB Order: GRB
[14:15:05][C][esp32_rmt_led_strip:282]:   Max refresh rate: 0
[14:15:05][C][light:092]: Light 'Seeed MR60BHA2 RGB Light'
[14:15:05][C][light:094]:   Default Transition Length: 1.0s
[14:15:05][C][light:095]:   Gamma Correct: 2.80
[14:15:05][C][bh1750.sensor:118]: BH1750 'Seeed MR60BHA2 Illuminance'
[14:15:05][C][bh1750.sensor:118]:   Device Class: 'illuminance'
[14:15:05][C][bh1750.sensor:118]:   State Class: 'measurement'
[14:15:05][C][bh1750.sensor:118]:   Unit of Measurement: 'lx'
[14:15:05][C][bh1750.sensor:118]:   Accuracy Decimals: 1
[14:15:05][C][bh1750.sensor:119]:   Address: 0x23
[14:15:05][C][bh1750.sensor:124]:   Update Interval: 1.0s
[14:15:05][C][restart.button:017]: Restart Button 'Restart'
[14:15:05][C][restart.button:017]:   Icon: 'mdi:restart'
[14:15:05][C][captive_portal:089]: Captive Portal:
[14:15:05][C][web_server:285]: Web Server:
[14:15:05][C][web_server:286]:   Address: 192.168.xxx.xxx:80
[14:15:05][C][mdns:116]: mDNS:
[14:15:05][C][mdns:117]:   Hostname: seeedstudio-mr60bha2-kit-b4b994
[14:15:05][C][esphome.ota:073]: Over-The-Air updates:
[14:15:05][C][esphome.ota:074]:   Address: 192.168.xxx.xxx:3232
[14:15:05][C][esphome.ota:075]:   Version: 2
[14:15:05][C][safe_mode:018]: Safe Mode:
[14:15:05][C][safe_mode:019]:   Boot considered successful after 60 seconds
[14:15:05][C][safe_mode:021]:   Invoke after 10 boot attempts
[14:15:05][C][safe_mode:022]:   Remain in safe mode for 300 seconds
[14:15:05][C][api:140]: API Server:
[14:15:05][C][api:141]:   Address: 192.168.xxx.xxx:6053
[14:15:05][C][api:145]:   Using noise encryption: NO
[14:15:05][C][seeed_mr60bha2:015]: MR60BHA2:
[14:15:05][C][seeed_mr60bha2:017]:  People Exist Binary Sensor 'Person Information'
[14:15:05][C][seeed_mr60bha2:017]:    Device Class: 'occupancy'
[14:15:05][C][seeed_mr60bha2:020]:  Breath Rate Sensor 'Real-time respiratory rate'
[14:15:05][C][seeed_mr60bha2:020]:    State Class: 'measurement'
[14:15:05][C][seeed_mr60bha2:020]:    Unit of Measurement: ''
[14:15:05][C][seeed_mr60bha2:020]:    Accuracy Decimals: 0
[14:15:05][C][seeed_mr60bha2:020]:    Icon: 'mdi:pulse'
[14:15:05][C][seeed_mr60bha2:023]:  Target Number Sensor 'Target number'
[14:15:05][C][seeed_mr60bha2:023]:    State Class: ''
[14:15:05][C][seeed_mr60bha2:023]:    Unit of Measurement: ''
[14:15:06][C][seeed_mr60bha2:023]:    Accuracy Decimals: 0
[14:15:06][D][bh1750.sensor:159]: 'Seeed MR60BHA2 Illuminance': Got illuminance=0.0lx
[14:15:06][D][sensor:093]: 'Seeed MR60BHA2 Illuminance': Sending state 0.00000 lx with 1 decimals of accuracy
[14:15:07][D][bh1750.sensor:159]: 'Seeed MR60BHA2 Illuminance': Got illuminance=0.0lx
[14:15:07][D][sensor:093]: 'Seeed MR60BHA2 Illuminance': Sending state 0.00000 lx with 1 decimals of accuracy
[14:15:08][D][bh1750.sensor:159]: 'Seeed MR60BHA2 Illuminance': Got illuminance=0.0lx
[14:15:08][D][sensor:093]: 'Seeed MR60BHA2 Illuminance': Sending state 0.00000 lx with 1 decimals of accuracy
[14:15:09][D][bh1750.sensor:159]: 'Seeed MR60BHA2 Illuminance': Got illuminance=0.0lx
[14:15:09][D][sensor:093]: 'Seeed MR60BHA2 Illuminance': Sending state 0.00000 lx with 1 decimals of accuracy
[14:15:10][D][bh1750.sensor:159]: 'Seeed MR60BHA2 Illuminance': Got illuminance=0.0lx
[14:15:10][D][sensor:093]: 'Seeed MR60BHA2 Illuminance': Sending state 0.00000 lx with 1 decimals of accuracy
[14:15:11][D][bh1750.sensor:159]: 'Seeed MR60BHA2 Illuminance': Got illuminance=0.0lx
[14:15:11][D][sensor:093]: 'Seeed MR60BHA2 Illuminance': Sending state 0.00000 lx with 1 decimals of accuracy
[14:15:12][D][bh1750.sensor:159]: 'Seeed MR60BHA2 Illuminance': Got illuminance=0.0lx
[14:15:12][D][sensor:093]: 'Seeed MR60BHA2 Illuminance': Sending state 0.00000 lx with 1 decimals of accuracy
[14:15:13][D][bh1750.sensor:159]: 'Seeed MR60BHA2 Illuminance': Got illuminance=0.0lx
[14:15:13][D][sensor:093]: 'Seeed MR60BHA2 Illuminance': Sending state 0.00000 lx with 1 decimals of accuracy
[14:15:14][D][bh1750.sensor:159]: 'Seeed MR60BHA2 Illuminance': Got illuminance=0.0lx
[14:15:14][D][sensor:093]: 'Seeed MR60BHA2 Illuminance': Sending state 0.00000 lx with 1 decimals of accuracy
[14:15:15][D][bh1750.sensor:159]: 'Seeed MR60BHA2 Illuminance': Got illuminance=0.0lx
[14:15:15][D][sensor:093]: 'Seeed MR60BHA2 Illuminance': Sending state 0.00000 lx with 1 decimals of accuracy
[14:15:16][D][bh1750.sensor:159]: 'Seeed MR60BHA2 Illuminance': Got illuminance=0.0lx
[14:15:16][D][sensor:093]: 'Seeed MR60BHA2 Illuminance': Sending state 0.00000 lx with 1 decimals of accuracy
[14:15:17][D][bh1750.sensor:159]: 'Seeed MR60BHA2 Illuminance': Got illuminance=0.0lx
[14:15:17][D][sensor:093]: 'Seeed MR60BHA2 Illuminance': Sending state 0.00000 lx with 1 decimals of accuracy
[14:15:18][D][bh1750.sensor:159]: 'Seeed MR60BHA2 Illuminance': Got illuminance=0.0lx
[14:15:18][D][sensor:093]: 'Seeed MR60BHA2 Illuminance': Sending state 0.00000 lx with 1 decimals of accuracy
[14:15:19][D][bh1750.sensor:159]: 'Seeed MR60BHA2 Illuminance': Got illuminance=0.0lx
[14:15:19][D][sensor:093]: 'Seeed MR60BHA2 Illuminance': Sending state 0.00000 lx with 1 decimals of accuracy

limengdu, 2025-05-06:
Please pay attention to the version number of your yaml, you are still using the outdated 2.0 version.
https://github.com/limengdu/MR60BHA2_ESPHome_external_components/blob/f875eba14198a61f13d39182ba6c2558b7401511/example/mr60bha2.yaml#L11

closed 2025-05-06 by limengdu

rickt123, 2025-05-06:
Sorry for the oversight.. will test out version 3.5