limengdu/MR60BHA2_ESPHome_external_components #19: How to Customize Detection Zones?

GuyFromMars365, 2025-05-04:
I noticed a Seeed blog from 2 months ago mentioned customizable "detection zones." Where is the tutorial for how to use that feature? I purchased these as an open-source alternative to the Aqara FP2, but I have yet to find a way to see & set zones. 

To be clear, I am looking for a GUI based way to view live output & adjust zones. As examples, both the [Aqara FP2](https://www.aqara.com/us/product/presence-sensor-fp2/) & [Everything Presence Lite](https://everythingsmarthome.github.io/everything-presence-lite/Home%20Assistant/creating-zones.html) have this function. I initially wanted the Everything Presence Light, but selected MR60BHA2 because I thought 60Ghz was more fine-tuned sensitive, the device looks smaller & it still gives me simple Home Assistant intergration with ability to customize. The inability to visually see live data & set zones is a major problem & providing thise function would greatly improve troubleshooting problems like interference or objects in room that creat false positives.

[image]

limengdu, 2025-05-06:
Sorry, the description here might not be comprehensive enough, causing you unnecessary trouble during use. This radar does support custom area detection settings, but it requires business cooperation to customize the radar. Currently, the relevant area setting interface has not been made available to users.

closed 2025-05-06 by limengdu

GuyFromMars365, 2025-05-07:
@limengdu I have emailed seeed support. Seeed needs to immediately correct & clairify their product listing & Wiki, as well as provide me a full refund, as this is false & misleading advertisement. Not saying it's intentional, but mentioning things like ESPhome & Home Assistant clearly appeals to DIY power users & mentioning things like "zones" is very clearly what other mmwave sensors offer, so to use all rhe buzzwords with none of the functionality (Zigbee & Matter are thrown around, but not implemented) is false & misleading.